hybridgroup/cylon #252: Support for Parrot MiniDrones and Bebop

AgileAce, 2015-01-02:
Hi. Any plans for adding support for the Parrot MiniDrones (Jumping Sumo and Rolling Spider) and Bebop? Parrot has recently released the latest SDK, which includes these devices: https://github.com/ARDroneSDK3/ARSDKBuildUtils.

deadprogram, 2015-01-03:
We've been planning on it for a while, now that the SDK is available, it becomes more realistic!

deadprogram, 2015-04-07:
Hi @AgileAce please checkout https://github.com/ChrisTheBaron/cylon-rolling-spider. 

We still need some wonderful community member like @ChrisTheBaron to implement Jumping Sumo and Bebot support.

ChrisTheBaron, 2015-04-07:
Hey @AgileAce,

Unfortunately at the moment I don't have either the Jumping Sumo or a Bebot, but if the demand is great enough then I might be tempted to purchase them and develop something for them.

Cheers, Chris

deadprogram, 2015-05-17:
For anyone wanting Parrot Rolling Spider support, @ChrisTheBaron created https://github.com/ChrisTheBaron/cylon-rolling-spider

For anyone wanting Parrot Bebop support, we ourselves created https://github.com/hybridgroup/cylon-bebop

Closing this issue... happy flying!

closed 2015-05-17 by deadprogram