hybridgroup/cylon #256: Problem with mqtt blink example

maqdes, 2015-01-13:
```javascript
"use strict";

var Cylon = require("cylon");

Cylon.robot({
  connections: {
    mqtt: { adaptor: "mqtt", host: "mqtt://localhost:1883" },
    firmata: { adaptor: "firmata", port: "COM5" }
  },

  devices: {
    toggle: { driver: "mqtt", topic: "toggle", adaptor: "mqtt" },
    led: { driver: "led", pin: "13", adaptor: "firmata" },
  },

  work: function(my) {
    my.toggle.on("message", function(data) {
      console.log("Message on 'toggle': " + data);
      my.led.toggle();
    });

    every((1).second(), function() {
      console.log("Toggling LED.");
      my.toggle.publish("toggle");
    });
  }
}).start();
```
# Result:

d:\Test\firmata\node_modules\cylon-gpio\lib\led.js:52
  this.connection.digitalWrite(this.pin, 1);
                  ^
TypeError: Object #<Adaptor> has no method 'digitalWrite'
    at Led.turnOn (d:\Test\firmata\node_modules\cylon-gpio\lib\led.js:52:19)
    at Led.toggle (d:\Test\firmata\node_modules\cylon-gpio\lib\led.js:71:10)
    at Driver.<anonymous> (d:\Test\firmata\server.js:19:14)
    at Driver.emit (events.js:95:17)
    at Driver.<anonymous> (d:\Test\firmata\node_modules\cylon-mqtt\lib\driver.js:30:12)
    at Adaptor.emit (events.js:98:17)
    at Adaptor.<anonymous> (d:\Test\firmata\node_modules\cylon-mqtt\lib\adaptor.js:28:10)
    at MqttClient.emit (events.js:106:17)
    at MqttClient._handlePublish (d:\Test\firmata\node_modules\cylon-mqtt\node_modules\mqtt\lib\client.js:544:12)
    at Connection.<anonymous> (d:\Test\firmata\node_modules\cylon-mqtt\node_modules\mqtt\lib\client.js:169:10)

stewart, 2015-01-13:
Sorry, that example was incorrect for some reason - this one should work:

```javascript
"use strict";

var Cylon = require("cylon");

Cylon.robot({
  connections: {
    mqtt: { adaptor: "mqtt", host: "mqtt://localhost:1883" },
    firmata: { adaptor: "firmata", port: "/dev/ttyACM0" }
  },

  devices: {
    toggle: { driver: "mqtt", topic: "toggle", connection: "mqtt" },
    led: { driver: "led", pin: "13", connection: "firmata" },
  },

  work: function(my) {
    my.toggle.on("message", function(data) {
      console.log("Message on 'toggle': " + data);
      my.led.toggle();
    });

    every((1).second(), function() {
      console.log("Toggling LED.");
      my.toggle.publish("toggle");
    });
  }
}).start();
```

The relevant example should be updated in-repo and on the docs site shortly.

Thanks for bringing this to our attention!

maqdes, 2015-01-13:
Thank you for quick response. Now everything is working.

closed 2015-01-13 by stewart