hybridgroup/cylon #259: API Not working in 0.22

lorenwest, 2015-01-19:
The API was split out for v0.22, so I'm getting this when I try using the API:
```
E, [2015-01-19T07:27:05.554Z] ERROR -- : The HTTP API is no longer included in Cylon by default.
E, [2015-01-19T07:27:05.557Z] ERROR -- : To use it, install the plugin module: `npm install cylon-api-http`
```

That's a helpful error message, but unfortunately `npm install cylon-api-http` gives an error because it hasn't yet been published.

Hopefully publishing `cylon-api-http` to npm will resolve the issue.

deadprogram, 2015-01-19:
Oops! We'll publish shortly, sorry about that.

deadprogram, 2015-01-19:
@lorenwest we've published it. Please give another try now.

lorenwest, 2015-01-19:
Thanks - that did the trick!  

My 0.22 installation got further, but failed when loading `cylon-gpio` because it couldn't find the `async` package, and async wasn't listed as a dependency in the `cylon-gpio/package.json` file.

lorenwest, 2015-01-19:
Looks like I had an older version of cylon-gpio.  No longer an issue.

closed 2015-01-19 by lorenwest