hybridgroup/cylon #260: Cylon API - events.js throw er

tomasinouk, 2015-01-22:
Hi, I am having strange behaviour running *Cylon API* on **Beaglebone Black**.
Script runs on Beaglebone Black itself.
BeagleBone Black runs *Cloud9 IDE* on port:`3000`.

When I use API to control `Firmata` devices it works as expected.

*Console output:*
```sh
root@beaglebone:/var/lib/cloud9/examples/cylon# node app.js 
I, [2015-01-22T02:30:04.503Z]  INFO -- : Initializing connections.
I, [2015-01-22T02:30:08.174Z]  INFO -- : Initializing devices.
I, [2015-01-22T02:30:08.264Z]  INFO -- : Starting connections.
I, [2015-01-22T02:30:08.294Z]  INFO -- : Starting devices.
I, [2015-01-22T02:30:08.319Z]  INFO -- : Working.
beagle says relax
running app.js
Cylon API Server is now online.
Listening at https://127.0.0.1:4321

events.js:72
        throw er; // Unhandled 'error' event
              ^
Error: listen EADDRINUSE
    at errnoException (net.js:904:11)
    at Server._listen2 (net.js:1042:14)
    at listen (net.js:1064:10)
    at net.js:1146:9
    at dns.js:72:18
    at process._tickCallback (node.js:419:13)
    at Function.Module.runMain (module.js:499:11)
    at startup (node.js:119:16)
    at node.js:906:3
```
**here is the script:**
file: `beaglebone_commands.js`

```
var Cylon = require('cylon');

function MyBeagle() {

var self = this;
Cylon.api(
  {
  port: '4321',
  }
);
Cylon.api();

Cylon.robot({
  name: "beagle",

  connections: {
    beaglebone: { 
      name: 'beagle',
    	adaptor: 'beaglebone', 
    },

  },

  devices: {
    led_blue: { driver: 'led', pin: 'P9_16' },
    led_red: {driver: 'led', pin: 'P9_14'},
    button: {driver: 'button', pin: 'P8_19'},
    led_orange: {driver: 'led', pin: 'P9_17'}
  },

  work: function(my)  {
    // every((0.2).second(), my.led.toggle);
    // every((0.2).second(), my.led_red.toggle);
    console.log( this.name + " says relax");
  },

   commands: function() {
    var commands = {};

    commands.do_a_thing = this.doAThing;
    commands.blink_red = this.blinkRed;
    commands.say_relax = this.sayRelax;

    return commands;
  },

  doAThing: function() {
    console.log("I did a thing!");
  },

    blinkRed: function() {
      this.led_red.toggle();
    // every((0.2).second(), this.led_red.toggle);
    console.log('blinkred');
  },
  sayRelax: function() {
   console.log( this.name + " says relax");
  }

}).start();

}

MyBeagle.prototype.flashRed = function() {
  console.log('flashing Red');
  setInterval(function() {
    Cylon.robots.beagle.devices.led_red.toggle();
  },1000);
  
};

MyBeagle.prototype.sayHi = function() {
  console.log('Aloha...');
  // setInterval(function() {
  //   Cylon.robots.beagle.devices.led_red.toggle();
  // },1000);
  
};

// Export the module
module.exports = exports = new MyBeagle();
```
file: `app.js`
```
var mybeagle = require('./beaglebone_commands');

var alarm = false;

setTimeout(function() {
	alarm = true;
},15000);

console.log('running app.js');
mybeagle.sayHi();

if (alarm) {
	mybeagle.flashRed();
	console.log('alarm - flashing red');
}
```

stewart, 2015-01-22:
Hi @tomasinouk!

From what I can see, the problem is that you have two calls to `Cylon#api`.

The first one is correctly starting on port `4321`, but the second call (`Cylon.api()`) is attempting to spin up another API instance, on the default port `3000`. This is colliding with your Cloud9 instance, resulting in the error. Removing that second call should fix your issue.

tomasinouk, 2015-01-22:
HI @stewart 
**you are right!**
I did not understood properly at first after reading the doco. 
That fixed my problem. 
Thanks a lot.

closed 2015-01-22 by tomasinouk