hybridgroup/cylon #263: serialport not compiling properly

ryantuck, 2015-02-04:
I'm trying to get cylon to work with node-webkit to talk to my arduino. 

When I try to run node-webkit, I get the followingn error:

[1044:0204/170108:INFO:CONSOLE(27)] ""It looks like serialport didn't compile properly. This is a common problem and its fix is well documented here https://github.com/voodootikigod/node-serialport#to-install"", source: /Users/ryan/Documents/active/panels/node_modules/cylon-firmata/node_modules/firmata/lib/firmata.js (27)

Unfortunately, the fix is not documented at all at the github link provided. 

Any suggestions on how to fix?

stewart, 2015-02-18:
Hi @ryantuck - sorry for not reporting back earlier. This is a known existing issue with node-webkit, the discussion here may prove helpful -> https://github.com/voodootikigod/node-serialport/issues/374.

edgarsilva, 2015-03-06:
Closing this issue since it is a problem with serialport and node-webkit not cylon itself.

closed 2015-03-06 by edgarsilva