hybridgroup/cylon #266: cyclon socket IO demo error net::ERR_CONNECTION_REFUSED on client

jBachalo, 2015-02-17:
Hi. 
Following this demo here exactly. The only difference is that I am using an edison instead of arduino.

http://cylonjs.com/blog/2015/02/05/using-socketio-with-cylonjs-api/

This is my server code.

    var Cylon = require('cylon');
     Cylon.robot({
        name: 'chappie',
        connections: {
            edison: { adaptor: 'intel-iot' }//,
            //led: { driver: 'led', pin: 13 }
      },
    
       devices: {
            maxbotix: { driver: 'maxbotix', pin: '0' },
            led: { driver: 'led', pin: 13 }
        },
        work: function (my) {
        /*
        STUFF TO DO!
      */
        }    
        })  
    Cylon.api(
    'socketio',
    {
        host: '0.0.0.0',//Is this correct??
        port: '3000'
     });

    Cylon.start();

The 'blink-client' is exactly the same as in the demo except I am hardcoding the actual IP to the edison where we define the device. ie

    device = io('http://<ip of edison>:3000/api/robots/chappie/devices/led');

When I runn app.js, the socketIO server correctly launches and I see

    [2015-02-17T20:04:12.813Z]  INFO -- : Initializing connections.
    [2015-02-17T20:04:13.116Z]  INFO -- : Initializing devices.
     [2015-02-17T20:04:14.646Z]  INFO -- : Starting connections.
    [2015-02-17T20:04:14.650Z]  INFO -- : Starting devices.
    [2015-02-17T20:04:14.667Z]  INFO -- : Working.
    [2015-02-17T20:04:14.703Z]  INFO -- : Cylon API Server is now online.
    [2015-02-17T20:04:14.704Z]  INFO -- : Listening at https://127.0.0.1:3000

But when I try and run the blink-client.html in a browser on a PC on the same newtwork  I get the ERR_CONNECTION_REFUSED

I am using cylon-api-socketio version "0.2.2" and nodejs version 0.10.28

stewart, 2015-02-17:
Hi @jBachalo - it looks like your Edison setup has an older version of Cylon installed - you should be able to grab the latest versions of everything with `npm update`.

That log output is from an older version of Cylon that had the HTTP API bundled in.

A newer version should look something like this when run:

```
$ node script
I, [2015-02-17T20:45:42.773Z]  INFO -- : Initializing connections.
I, [2015-02-17T20:45:42.885Z]  INFO -- : Initializing devices.
Socket.io has started...
I, [2015-02-17T20:45:43.315Z]  INFO -- : Starting connections.
Cylon + Socket.io is now online.
Listening Cylon Socket.io API Server @0.0.0.0:3000
I, [2015-02-17T20:45:43.319Z]  INFO -- : Starting devices.
I, [2015-02-17T20:45:43.319Z]  INFO -- : Working.
```

jBachalo, 2015-02-17:
Thanks Andrew. Will try that first

jBachalo, 2015-02-17:
Hi Andrew
Ran npm update but still looks the same when I start the server. Is it
possible I need to correct my package.json file BEFORE I run npm update?

Right now I'm seeing the following in my package.json

"dependencies": {
    "cylon": "^0.21.2",
    "cylon-api-socketio": "^0.2.2",
    "cylon-gpio": "^0.21.0",
    "cylon-i2c": "^0.18.0",
    "cylon-intel-iot": "^0.4.0"

stewart, 2015-02-17:
Hi, sorry, should have led with that:

`dependencies` section should look more like this:

```
"dependencies": {
  "cylon": "^0.22.0",
  "cylon-api-socketio": "^0.2.2",
  "cylon-gpio": "^0.22.0",
  "cylon-i2c": "^0.19.0",
  "cylon-intel-iot": "^0.5.0"
}
```

jBachalo, 2015-02-17:
thanks. working! How would I add a socket.emit or broadcast on the server
side?

edgarsilva, 2015-02-17:
@jBachalo you can check this example here:

https://github.com/hybridgroup/cylon-api-socketio/blob/master/examples/robot_events_commands/robot-events-commands-server.js

You need to register events on your robot for the socketio API to pick them up like this:

```javascript
Cylon.robot({
  name: 'cybot',
  // These are the events that will be registered in the API
  events: ['turned-on', 'turned-off'],
  // .... more code ...
});

```

After that you just emit events on your robot like this:

```javascript
turnOn: function() {
  this.led.turnOn();
  this.emit('turned-on');
}
```

All the events registered in the robot would be available in the socket client, and you can listen for them doing something like:

```javascript
robot = io('http://127.0.0.1:3000/api/robots/cybot');

robot.on('turned-on', function(){
  console.log('THE ROBOT JUST TURNED ON ITS LED!!!');
});
```

You can check the full examples here:

https://github.com/hybridgroup/cylon-api-socketio/tree/master/examples/robot_events_commands

jBachalo, 2015-02-17:
great thanks again

closed 2015-02-18 by stewart