hybridgroup/cylon #267: TypeError: Cannot read property 'motor' of undefined

Evoliofly, 2015-02-19:
Hello, I encountered an issue while trying to run a demo from the doc.
I try to start and stop a motor on an arduino with firmata from a Raspberry Pi. 
First of all, uname:
Linux raspberrypi 3.12.35+ #730 PREEMPT Fri Dec 19 18:31:24 GMT 2014 armv6l GNU/Linux
and node v0.10.36

Here's what I'm running:

'''var Cylon = require('cylon');

Cylon.robot({
  connections: {
    arduino: { adaptor: 'firmata', port: '/dev/ttyACM0' }
  },

  devices: {
    motor: { driver: 'motor', pin: 3 }
  },

  work: function(my) {
    my.motor.speed(100);

    after((1).seconds(), function(my) {
      my.motor.stop();
    });
  }
}).start();'''

'''pi@raspberrypi ~ $ node ppe3.js
I, [2015-02-19T01:03:42.253Z]  INFO -- : Initializing connections.
I, [2015-02-19T01:03:44.363Z]  INFO -- : Initializing devices.
I, [2015-02-19T01:03:44.493Z]  INFO -- : Starting connections.
I, [2015-02-19T01:03:47.734Z]  INFO -- : Starting devices.
I, [2015-02-19T01:03:47.743Z]  INFO -- : Working.

/home/pi/ppe3.js:16
      my.motor.stop();
        ^
TypeError: Cannot read property 'motor' of undefined
    at null._onTimeout (/home/pi/ppe3.js:16:9)
    at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)'''

Also...

'''pi@raspberrypi ~ $ npm ls
/home/pi
├─┬ cylon@0.22.0
│ └── async@0.9.0
├─┬ cylon-firmata@0.19.0
│ ├─┬ cylon@0.22.0
│ │ └── async@0.9.0
│ ├── cylon-gpio@0.22.0
│ ├── cylon-i2c@0.19.0
│ └─┬ firmata@0.4.0
│   ├── browser-serialport@2.0.1
│   ├── object-assign@1.0.0
│   └─┬ serialport@1.4.10
│     ├── async@0.9.0
│     ├── bindings@1.2.1
│     ├── nan@1.3.0
│     ├─┬ node-pre-gyp@0.5.19
│     │ ├─┬ mkdirp@0.5.0
│     │ │ └── minimist@0.0.8
│     │ ├─┬ nopt@2.2.1
│     │ │ └── abbrev@1.0.5
│     │ ├─┬ npmlog@0.0.6
│     │ │ └── ansi@0.2.1
│     │ ├─┬ rc@0.4.0
│     │ │ ├── deep-extend@0.2.10
│     │ │ ├── ini@1.1.0
[... 262 lines cut ...]
  │ ├── uid-number@0.0.6
  │ ├── umask@1.1.0
  │ ├── which@1.0.8
  │ ├── wrappy@1.0.1
  │ └── write-file-atomic@1.1.0
  ├─┬ optimist@0.6.1
  │ ├── minimist@0.0.10
  │ └── wordwrap@0.0.2
  ├── rimraf@2.2.8
  └─┬ tar.gz@0.1.1
    ├─┬ commander@1.1.1
    │ └── keypress@0.1.0
    ├─┬ fstream@0.1.31
    │ ├── graceful-fs@3.0.5
    │ ├── inherits@2.0.1
    │ └─┬ mkdirp@0.5.0
    │   └── minimist@0.0.8
    └─┬ tar@0.1.20
      ├── block-stream@0.0.7
      └── inherits@2.0.1
'''

So, the motor is working but I don't manage to stop it...

Thank you. :)

stewart, 2015-02-19:
Hi @Evoliofly - in your code you're passing introducing another `my` scope here:

```javascript
    after((1).seconds(), function(my) { // <- here
      my.motor.stop();
    });
```

If you get rid of that extra `my`, your code should work.

Evoliofly, 2015-02-19:
Thank you.
( http://cylonjs.com/documentation/drivers/motor/ )

closed 2015-02-19 by Evoliofly