hybridgroup/cylon #273: i2c version of the pca9685

cybertim, 2015-02-28:
I would like to request a driver for the i2c version of the pca9685 (available as a module for Arduino and Raspberry Pi).
Currently there is only a driver for the Tessel pca9685 which doesn't seem to be compatible with the Arduino and Raspberry Pi device.

deadprogram, 2015-02-28:
Hi, @cybertim 

I do not have this servo controller, but I put together some preliminary support that *might* work here https://github.com/hybridgroup/cylon-i2c/tree/feature/pca9685

Can you please pull that branch down locally, and see if it works for you? Please let us know what happens.

cybertim, 2015-03-01:
Hi and thanks for the quick follow up.

I've been tinkering with your branch half a day now, but there seems to be some issues.

First of all, I found this library:
https://github.com/johntreacy/adafruit-pca9685/blob/master/adafruit-pca9685.js
Which seems to work with the i2c pca9684 on a raspberry pi. But it isn't in Cylon ;-)

I've managed to get some results (the servo's do seem to get a quick power surge now instead of nothing happening) by adding a different 'sleep' (like in the adafruit-pca9685.js) function when setting the mode. The Cylon.Utils.sleep doesn't seem to work(?)

But even after copy pasting most of the stuff from the adafruit-pca9685.js into the 'i2cWrite and i2cRead' functions of cylon, the servo's refuse to move. They do get a quick 'shot' on the first try but after this I need to restart the whole pca9685 board.. (crashed?) This issue does not happen when using the adafruit-pca9685 lib.

I will probably try some more tinkering when I find the time. Hope this information can be useful in the meantime.

cybertim, 2015-03-02:
Ok by making a typo in some test code I figured out that the 'i2cRead' function is doing something wrong.
So the value read from 0x00 is wrong and writing it back when setting the frequency kills the board.
The working code uses 'i2c.readBytes' from "i2c": "~0.1.4".
Is this comparable with the i2cRead function from cylon?

cybertim, 2015-03-03:
Found the issue, but I think it is related to the i2c implementation of the raspi. I didn't have the time yet to check out all the libraries and trace it back. But maybe this can be of assistance to fix the issue asap:

The cylon-raspi probably implements i2cRead and i2cWrite.
These functions probably directly implement the i2c.readBytes and i2c.writeBytes.
This is okay, but for some reason all the current implementations of cylon i2c chips call the i2cRead and write and expect a single value back.
But in reality (on the raspi) it is an Array (!!)
Also writing to the i2c needs an Array.

Maybe this is also the issue with other issues reported about the i2c on the raspi.

Check out my WORKING version of the 'driver' here in gist: https://gist.github.com/cybertim/bd77aa728a3f9dc3ef87

edgarsilva, 2015-03-03:
I think you are doing some stuff that should not be there in the driver and the adaptor already takes care of, let me refactor your code a bit and send it back, maybe that will put you back on track.

edgarsilva, 2015-03-03:
can you try something like this for your driver pls and let me know?

I'm not sure if that works, I do not have the hardware to test it, but you can see there the way it should be implemented. A few changes in the way you call i2cRead and i2cWrite, as well as returning
the `oldmode` value, that should be inside the `data` param passed to the callback on reads.

```javascript
/*
 * PCA9685 driver
 * http://cylonjs.com
 *
 * Copyright (c) 2015 The Hybrid Group
 * Licensed under the Apache 2.0 license.
 */

// jshint unused:false

"use strict";

var Cylon = require("cylon");
var sleep = require('sleep');

var PCA9685_MODE1 = 0x00,
PCA9685_PRESCALE = 0xFE,
PCA9685_SUBADR1 = 0x02,
PCA9685_SUBADR2 = 0x03,
PCA9685_SUBADR3 = 0x04,
LED0_ON_L = 0x06,
LED0_ON_H = 0x07,
LED0_OFF_L = 0x08,
LED0_OFF_H = 0x09,
ALLLED_ON_L = 0xFA,
ALLLED_ON_H = 0xFB,
ALLLED_OFF_L = 0xFC,
ALLLED_OFF_H = 0xFD;

/**
 * PCA9685 Driver
 *
 * @constructor pca9685
 */
var PCA9685 = module.exports = function PCA9685() {
  PCA9685.__super__.constructor.apply(this, arguments);
  this.address = 0x40;
  this.commands = {
    reset: this.reset,
    stop: this.stop,
    set_pwm_freq: this.setPWMFreq,
    set_pwm: this.setPWM
  };
};

Cylon.Utils.subclass(PCA9685, Cylon.Driver);

/**
 * Starts the driver
 *
 * @param {Function} callback triggered when the driver is started
 * @return {null}
 */
PCA9685.prototype.start = function(callback) {
  this.i2cSend(PCA9685_MODE1, 0x00);
  this.i2cSend(ALLLED_OFF_H, 0x10);
  callback();
};

/**
 * Stops the driver
 *
 * @param {Function} callback triggered when the driver is halted
 * @return {null}
 */
PCA9685.prototype.halt = function(callback) {
  callback();
};

// THIS SHOULD NOT BE NECCESSARY, IS THERE A REASON FOR IT?
/**
 * Resets the PCA9685
 * @return {null}
 */
//PCA9685.prototype.reset = function(callback) {
  //callback();
//};

PCA9685.prototype.stop = function() {
  this.i2cSend(ALLLED_OFF_H, 0x10);
};

/**
 * Set the servo frequency for the PCA9685
 *
 * @param {number} frequency
 * @return {null}
 * @publish
 */
PCA9685.prototype.setPWMFreq = function(frequency) {
  // Adjust per https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/
  //frequency *= 0.9;
  var prescaleval = 25000000;
  prescaleval /= 4096.0;
  prescaleval /= frequency;
  prescaleval -= 1.0;
  var prescale = Math.floor(prescaleval * 1.0 + 0.5);

  this.i2cRead(this.address, PCA9685_MODE1, 1, function(err, data) {
    // THIS CALLBACK GETS THE DATA  FROM THE I2C READ
    // IT SHOULD CONTAIN WHATEVER YOU READ FROM I2C
    // USE THAT TO GET THE OLDMODE
    console.log("OLDMODE ====>", data);
    var oldmode = data[0]; // NOT SURE HOW THIS VALUE IS RETURNED

    // MAKE SURE NEWMODE OR THE VALUES PASSED TO WRITE IS AN ARRAY
    // OR A BUFFER
    var newmode = [(oldmode & 0x7F)] | [0x10]; // sleep

    this.i2cWrite(this.address, PCA9685_MODE1, newmode);
    this.i2cWrite(this.address, PCA9685_PRESCALE, Math.floor(prescale));

    // SAME HERE YOU NEED TO PASS OLDMODE AS AN ARRAY
    this.i2cWrite(this.address, PCA9685_MODE1, [oldmode]);
    sleep.usleep(10000);
    this.i2cWrite(this.address, PCA9685_MODE1, [oldmode] | [0x80]);
  });
};

/**
 * Set the servo position for the PCA9685
 *
 * @param {number} servoNum
 * @param {number} on
 * @param {number} off
 * @return {null}
 * @publish
 */
PCA9685.prototype.setPWM = function(channel, pulseon, pulseoff) {
  this.i2cWrite(this.address, LED0_ON_L + 4 * channel, [pulseon & 0xFF]);
  this.i2cWrite(this.address, LED0_ON_H + 4 * channel, [pulseon >> 8]);
  this.i2cWrite(this.address, LED0_OFF_L + 4 * channel, [pulseoff & 0xFF]);
  this.i2cWrite(this.address, LED0_OFF_H + 4 * channel, [pulseoff >> 8]);
};

/* not nedded you should just send it to
 * connection.i2cRead and connection.i2cWrite
 * this way we make sure your driver works on
 * different platforms.
PCA9685.prototype.i2cRead = function(cmd, length) {
  var result;
  this.connection.i2cRead(this.address, cmd, length, function(err, res) {
    if (err != null) {
      console.log("I2C read error", err);
    }
    result = res[0];
  });
  return result;
};

PCA9685.prototype.i2cSend = function(cmd, values) {
  if (!(values instanceof Array))
    values = [values];
  this.connection.i2cWrite(this.address, cmd, values, function(error) {
    if (error != null)
      console.log("I2C error", error);
  });
}
*/ 
```

cybertim, 2015-03-03:
Did a quick test and altered it, now it's working!

Check the altered code here:
https://gist.github.com/cybertim/bd77aa728a3f9dc3ef87#comment-1405092

edgarsilva, 2015-03-03:
cool, you can fork the cylon-i2c repo and submit a PR when you've added some tests, I canc heck it out later today.

cybertim, 2015-03-06:
next problem is, how does cylon solve the dependencies of 'servo' and 'led' for the pca9685?

I looked into the servo function code, but they are heavily based on the GPIO direct connection, it's not possible to use some generic or abstract implementations of the for ex. setAngle function etc. and connect them to the i2c version of the  pca9685 driver..  (?)
If this is possible, which impl. can i use as an example?

edgarsilva, 2015-03-06:
Hi @cybertim, that is not possible, those drivers are only for GPIO. Since the implementation of this would me married to that I2C driver they sould be there in the driver itself.

Or create a generic implementation for I2C pwm drivers, which might be difficult since we do not know how every other I2C pwm driver implements reads and writes.

closed 2015-03-06 by cybertim