hybridgroup/cylon #274: Raspi I2C

al555bike, 2015-03-02:
Hi, Your Cylon libraries are great. Used the GPIO and PWM ones so far. I've started using the I2c one to control an LCD, but am having problems. I have followed all the instructions for changing the configuration files and am using your example below. It gives me the error 
TypeError: Failed to set address
    at i2c.setAddress (/home/pi/node_modules/cylon-raspi/node_modules/i2c/lib/i2c.coffee:61:12
and some similar lines. Can you advise me what I'm doing wrong please. 
Thanks 
Alastair

Cylon = require('cylon');
Cylon.robot({
  connections: {
    raspi: { adaptor: 'raspi', port: '/dev/ttyACM0' }
  },
  devices: {
    lcd: { driver: 'lcd' }
  },
  work: function(my) {
    my.lcd.on('start', function(){
      my.lcd.print("Hello!");
    });
  }
}).start()

edgarsilva, 2015-03-03:
What board are you using? Arduino? beaglebone black? Raspberry Pi?

edgarsilva, 2015-03-03:
Nevermind just read the addrees for the raspi, can you verify the lcd is being recognized using i2c-tools?

edgarsilva, 2015-03-03:
You can try connecting the LCD and doing something like:

```bash
$ i2cdetect -l
$ i2cdetect 0
```

Run it before and after connecting the LCD you should see the LCD address listed there.

al555bike, 2015-03-03:
Sorry should have said, it’s a Raspberry Pi B+

This email and any attachments are confidential and intended solely for the use of the addressee and may contain information which is covered by legal, professional or other privilege. If you have received this email in error please notify the system manager at postmaster@ulster.ac.uk and delete this email immediately. Any views or opinions expressed are solely those of the author and do not necessarily represent those of Ulster University. The University's computer systems may be monitored and communications carried out on them may be recorded to secure the effective operation of the system and for other lawful purposes. Ulster University does not guarantee that this email or any attachments are free from viruses or 100% secure. Unless expressly stated in the body of a separate attachment, the text of email is not intended to form a binding contract. Correspondence to and from the University may be subject to requests for disclosure by 3rd parties under relevant legislation. The University of Ulster was founded by Royal Charter in 1984 and is registered with company number RC000726 and VAT registered number GB672390524.The primary contact address for Ulster University in Northern Ireland is Cromore Road, Coleraine, Co. Londonderry BT52 1SA

al555bike, 2015-03-04:
Very many thanks Edgar, I got it working. Being new to Pi and Linux I hadn't installed i2c-tools or activated the i2c interface. All done now and working. Can you point me to some examples of using the drivers other than the basic one. Want to know how to change lines etc.
Thanks
Alastair

edgarsilva, 2015-03-04:
There are examples in the repos here:

https://github.com/hybridgroup/cylon-i2c/tree/master/examples
https://github.com/hybridgroup/cylon-firmata/tree/master/examples

And in the docs here:
http://cylonjs.com/documentation/examples/

al555bike, 2015-03-04:
Hi Edgar, yes I had found these resources. They show a simple "Hello" program. It's the details of the other methods within LCD that I would like details to. For instance the setCursor method. What data does it accept? How do you specify row and char position. Is there a link to details of how to use the various methods in LCD and other drivers. Some of them are self explanatory such as blink and cursorOff. Servo for instance shows how to pass parameters to safeAngle.

edgarsilva, 2015-03-04:
They show the basic usage for each of the I2C drivers, if you want to check other methods, events etc.. you need to look into the docs here:

http://cylonjs.com/documentation/

There is an I2C section for each driver, it contains commands, methods, events (where available), etc..

edgarsilva, 2015-03-06:
Closing this issue since it seems solved.

closed 2015-03-06 by edgarsilva