hybridgroup/cylon #277: Problems when running node.js with cylon.js as service in Intel Edison

X-Y, 2015-03-13:
I've made a node.js script to run a robot. The script itself runs fine, but if I want to use it as a service, then it always fails. The following code starts the problem:
 
***index.js***
```javascript
var app = require('express')();
var http = require('http').Server(app);
var io = require('socket.io')(http);
 
var Cylon = require('cylon');
 
//The following part kills the servive
var rbt=Cylon.robot({
  connections: {
    edison:{adaptor: 'intel-iot'}
  },
 
  devices: {
    servoL: { driver: 'servo', pin: 20 },
    servoR: { driver: 'servo', pin: 14 },
    button: { driver: 'button', pin: 15 },
  },
});
 ```
***/etc/systemd/system/robotserver.service***
```
[Unit]
Description=robotserver startup
After=network.service
 
[Service]
ExecStart=/bin/sh -c "/usr/bin/node /home/root/robotServer/index.js"
Restart=always
RestartSec=10s

[Install]
WantedBy=multi-user.target
 ```

And I've executed the following commands in terminal:
```
systemctl daemon-reload
systemctl start robotserver.service
```

If I run "systemctl status robotserver.service", the following message will show up:
```
● robotserver.service - robotserver startup
   Loaded: loaded (/etc/systemd/system/robotserver.service; disabled)
   Active: failed (Result: exit-code) since Fri 2015-03-13 09:14:39 UTC; 9s ago
  Process: 276 ExecStart=/usr/bin/node /home/root/robotServer/index.js (code=exited, status=143)
 Main PID: 276 (code=exited, status=143)

Mar 13 09:14:35 edison systemd[1]: Started robotserver startup.
Mar 13 09:14:38 edison node[276]: I, [2015-03-13T09:14:38.911Z]  INFO -- : [Robot 60097] - Initializing connections.
Mar 13 09:14:39 edison node[276]: Cannot find the 'cylon-intel-iot' module.
Mar 13 09:14:39 edison node[276]: This problem might be fixed by installing it with 'npm install cylon-intel-iot' and trying again.
Mar 13 09:14:39 edison systemd[1]: robotserver.service: main process exited, code=exited, status=143/n/a
Mar 13 09:14:39 edison systemd[1]: Unit robotserver.service entered failed state.
```
The script runs fine itself, so I don't understand why it's asking for cylon-intel-iot module when it's already installed.
Wondering if it's a bug or I did something wrong here?

The original post can be seen at https://communities.intel.com/thread/61262

zankich, 2015-03-15:
@X-Y This is a path issue related to your project. Do you have `cylon` installed as a global node module and `cylon-intel-iot` at a project level? I would think that if node can find `cylon` then it would also find `cylon-intel-iot`...

X-Y, 2015-03-16:
@zankich I have both of them installed as project level modules, it's indeed pretty strange...

deadprogram, 2015-05-26:
Hello @X-Y any update on this?

edgarsilva, 2015-06-24:
Maybe it has something to do with where `systemd` is running it from, I mean the path nodejs takes for the dependencies of the script being run is usually the same as where you are executing node from and not where the script resides.

So in this case if you install them globally so systemd can find them might work.

I just received and intel edison and I would give this a try today, in the meantime you could give it a try by installing them globally.

r2DoesInc, 2015-06-29:
http://stackoverflow.com/questions/31121914/node-dependencies-not-found-when-ran-programatically?noredirect=1#comment50259590_31121914

Ran into this issue myself and have not been able to figure it out yet.

r2DoesInc, 2015-06-30:
For me, the issue was in the package.json. Somehow it was empty.
Run ` require('cylon-intel-iot')` from REPL in your project and see if you get any errors.

edgarsilva, 2015-07-02:
@X-Y have you had any luck with this? it seems to be an issue with either your NODE_PATH or module installation. I just ran your example with the latest Edison firmware update with no issues starting, stopping and restarting the service.

Can you share a list of the `node_modules` inside the `index.js` tree. Also the global ones.

It also seems like you are trying to setup monitoring and controlling of your robot using socket-io. We have an API that does just that without you having to hook everything up.

You can check it out here:
http://cylonjs.com/documentation/api/socket-io/

What @r2DoesInc mentions is also correct, if you did not install manually each module, make sure to include all of them in you `package.json` and that you ran `npm install`. One last thing did you install `cylon-intel-iot` globally or just in your project?

edgarsilva, 2015-07-02:
@X-Y great! I was able to reproduce your exact issue by just installing all the modules in the global npm module folder. It seems systemd has some issues with the path for the global folder. You can easily fix this by following @r2DoesInc recomendation. Add a `package.json` to your project, with the appropriate modules, and run `npm install`. it should look something like this:

***./package.json***
```json
{
  "dependencies": {
    "cylon":      "~1.1.0",
    "cylon-gpio": "~0.26.0",
    "cylon-i2c":  "~0.22.0",
    "cylon-intel-iot": "~0.7.0"
  }
}
```

edgarsilva, 2015-07-02:
I'm gonna close this issue since this should fix it, if issue still persists pls reopen.

r2DoesInc, 2015-08-05:
I have been running into this problem again recently. 

to start the bot, I have even tried to start the server with 
`exec('npm install cylon-intel-iot && npm install && node server.js,...'`
which results in the following getting printed to my log
>cylon-intel-iot@0.6.0 node_modules/cylon-intel-iot
>â^Ô^Ôâ^Ô^Àâ^Ô^À cylon-gpio@0.23.0
>I, [2015-08-04T19:17:02.925Z]  INFO -- : [ardrobot] - Initializing connections.
>Cannot find the 'cylon-intel-iot' module.
>This problem might be fixed by installing it with 'npm install cylon-intel-iot' and trying again.
>
>npm WARN package.json ArdrobotEdison@1.0.0 No repository field.
>npm WARN package.json ArdrobotEdison@1.0.0 No README data
>npm WARN package.json ArdrobotEdison@1.0.0 No repository field.
>npm WARN package.json ArdrobotEdison@1.0.0 No README data

edgarsilva, 2015-08-06:
@r2DoesInc I'll check this in a couple of hours, I'm updating the edison firmware ATM. In the meantime could you pls share the `package.json` for your project and the directory tree for the `node_modules` folder?

r2DoesInc, 2015-08-06:
https://github.com/cloudspace/ArdrobotEdison/blob/master/package.json

edgarsilva, 2015-08-07:
Ok I think I found the problem, this code works for me:

```bash
[Unit]
Description=firealarm startup
After=network.service

[Service]
ExecStart=/usr/bin/node /home/root/workspace/fire_alarm/index.js
Restart=always
RestartSec=10s

[Install]
WantedBy=multi-user.target
```

The catch is that I had to install all npm dependencies + `mraa` in the project working directory. Previous to that I was seeing the `code=exited status=143` error you described above.

this is how my node_modules/npm instalation folder looks like now:

```bash
@ /home/root/workspace/fire_alarm
├── cylon@1.1.0
├── cylon-gpio@0.26.0
├── cylon-i2c@0.22.0
├── cylon-intel-iot@0.7.0
└── mraa@0.7.3 extraneous
```

@r2DoesInc Could you please give that a try and let me know if it also works for you?

r2DoesInc, 2015-08-10:
After struggling a bit with mraa, it seems to be working as expected now. This is quite odd, as mraa is only needed when running the project as a startup service. 

Ive added mraa to my projects dependency list, but is it actually one of cylons dependencies? Im not 100% sure where the issue actually lies.

edgarsilva, 2015-08-10:
@r2DoesInc after some testing, I've found this is happening because the global modules  path (where `mraa` usually resides) it is not being used by systemd, and only local dependencies are used in the project, that's what I've found so far, `mraa` is not needed when the project is not run as a service because it is usually installed globally in the edison, and by just doing `node index.js` in the project folder node has no issue finding the dependencies.

Regarding `mraa` being required by `cylon-intel-iot` also, this has to do on how npm implements dependencies and nested requirements. It is weird I agree, I think it would be better to have only one instance per module in the root folder, but I digress that is not how npm works.

Let me know if you have any more issues, I'll close this one now.

Regards,

closed 2015-08-10 by edgarsilva

trinitronx, 2016-06-27:
For Googlers:  @edgarsilva is correct. It seems that the environment SystemD starts the service in skips the global `/usr/lib/node_modules` folder, so the system-level `mraa` dependency is missing.

I was able to get this working by explicitly setting `NODE_PATH` and running `npm install` as a pre-exec step.  An example might be:

```
[Service]
Environment=NODE_ENV=production
Environment=NODE_PATH=/usr/lib/node_modules/:/home/root/.node_app_slot/node_modules
WorkingDirectory=/home/root/.node_app_slot/
ExecStartPre=-/usr/bin/npm install
ExecStart=/usr/bin/node /home/root/.node_app_slot/main.js
```