hybridgroup/cylon #288: robot data to http server instead of console.log?

rchongtay, 2015-04-16:
Hi,
Is it possible to have an example where the robot data for example from neurosky is displayed to  
http server instead of the console.log?

deadprogram, 2015-04-16:
Hi, @rchongtay 

You can do this right now using either the `cylon-api-http` with Server-Side Events (SSE) https://github.com/hybridgroup/cylon-api-http or else using `cylon-api-socketio` https://github.com/hybridgroup/cylon-api-socketio

Looks like we need an example that shows SSE in the simplest possible case. @stewart do you happen to know if we have one?

stewart, 2015-04-16:
Hi @rchongtay - based on my reading of your question, it looks more like you might be interested in a custom logger for Cylon.

Cylon supports custom logger implementations, and you can see an example of one in our [Browserify example][browserify].

You can find the custom logger implementation example [here][implementation], and how to add it to Cylon [here][usage].

If I've misread what you're looking for, please let us know.

[browserify]: https://github.com/hybridgroup/cylon-example-browserify
[implementation]: https://github.com/hybridgroup/cylon-example-browserify/blob/master/browser-logger.js
[usage]: https://github.com/hybridgroup/cylon-example-browserify/blob/master/script.js#L5-L15

rchongtay, 2015-04-16:
Hi, 
Thanks for your replies, I should be able to use one of these options, you understood well, at first I just want to display the data in the browser, then I will display it graphically instead, very much like the demonstration made in the video with Ron Evans (https://vimeo.com/96477741 min 20:11 and min 20:13 use of Logger.info and finally the graph in min 20:47).

I will try to follow the examples you refer to, thanks.

deadprogram, 2015-05-17:
Closing... please reopen if we can provide further assistance/

closed 2015-05-17 by deadprogram