hybridgroup/cylon #289: Some question about cylon.

iClosedz, 2015-04-21:
Hello!, I have watched this video demonstration (https://vimeo.com/96477741) about Cylon
and I think it not the same as Johnny-Five do. I have question to make me confirm that it doesn't like Johnny-Five.

When I run cylon (node myscript.js) it will upload the code to my Arduino ? So when me put my USB off the code also in my Arduino and when I give my Arduino the Power adapter my Arduino will run my old script (myscript.js) that right ?

iClosedz, 2015-04-21:
I got it. It can run on Tessel only.

closed 2015-04-21 by iClosedz

edgarsilva, 2015-04-21:
That is not the case, Cylon is a robotics framework that can run in more than 30 platforms, as compared to J5 which is more like a library for firmata compatible devices I believe.

You can check more about Cylon.js and all the platforms it supports here:

http://cylonjs.com/documentation/platforms/

Let us know if you have any more questions, but you can easily run on arduino, imp, spark, beaglebone black, raspberry-pi, digispark, Intel Galileo, Intel Edison... among others.