hybridgroup/cylon #291: Connections mixed up when using leapmotion with sphero

ddm, 2015-05-15:
The following code highlights the issue:

```
 require('cylon').robot({
  connections: {
    sphero: { adaptor: 'sphero', port: '/dev/tty.Sphero-PGP-AMP-SPP' },
    leapmotion: { adaptor: 'leapmotion' }
  },

  devices: {
    sphero: { driver: 'sphero' },
    leapmotion: { driver: 'leapmotion' }
  },

  work: function(my) {
    var leapMotionConnectionName = my.leapmotion.connection.name;
    console.log("\n# Expected leapmotion but got", leapMotionConnectionName, "#\n");
  }
}).start();
```

The output is:

```
I, [2015-05-15T10:29:37.951Z]  INFO -- : [Robot 57715] - Initializing connections.
I, [2015-05-15T10:29:38.019Z]  INFO -- : [Robot 57715] - Initializing devices.
I, [2015-05-15T10:29:38.024Z]  INFO -- : [Robot 57715] - Starting connections.
I, [2015-05-15T10:29:38.025Z]  INFO -- : Connecting to Sphero 'sphero'...
I, [2015-05-15T10:29:38.060Z]  INFO -- : [Robot 57715] - Starting devices.
I, [2015-05-15T10:29:38.063Z]  INFO -- : [Robot 57715] - Working.

# Expected leapmotion but got sphero #

Optimized for desktop usage.
```

ddm, 2015-05-15:
My bad! the correct way to do this would be

```
require('cylon').robot({
  connections: {
    sphero: {
      adaptor: 'sphero',
      port: '/dev/tty.Sphero-PGP-AMP-SPP',
      devices: {
        sphero: {
          driver: 'sphero'
        }
      }
    },
    leapmotion: {
      adaptor: 'leapmotion',
      devices: {
        leapmotion: {
          driver: 'leapmotion'
        }
      }
    }
  },

  work: function (my) {
    var leapMotionConnectionName = my.leapmotion.connection.name;
    console.log("\n# Expected leapmotion but got", leapMotionConnectionName, "#\n");
  }
}).start();
```

ddm, 2015-05-15:
Wrong again... Sorry for the noise but this is confusing :/

closed 2015-05-15 by ddm

ddm, 2015-05-15:
Closed because it works with the second example. I believe the documentation should reflect this because it encourages what I tried initially (i.e. not nesting devices under connections) which did not work...

edgarsilva, 2015-05-15:
Hey @ddm, You are correct passing the connection to the device if they are specefied in the last example also works. We should have some examples of this, here:

http://cylonjs.com/documentation/examples/sphero/js/pebble/
http://cylonjs.com/documentation/examples/force/js/sf-sphero/

But we need an easier way to point to those or include examples on how to do it in the core. @stewart do you know if we have examples of this or more complex code that are not hiding in the list of examples on cylon.js?

ddm, 2015-05-15:
I should have explored the examples before raising the issue. It seems perfectly logical, in retrospect, to specify the connection explicitly.
However the implicit use of the first connection when none was specified made the error I encountered confusing...

ddm, 2015-05-15:
That said I'm having tons of fun with Cylon :) Great job!

edgarsilva, 2015-05-15:
Thanks, we are working on improving the docs and as you already pointed out make errors more descriptive as well as adding a section to cylon.js on how to specify multiple connections and devices. To avoid confusion in the future.

Thanks for the feedback and feel free to close the issue, or we'll close it when the examples and updates have been deployed.

ddm, 2015-05-15:
I believe this was kind of a pebcak. I actually closed the issue when I
realised there was a built-in solution to my problem.

I was a bit too quick to stitch together the base examples of leapmotion
and sphero before reading the more complex examples ;)

edgarsilva, 2015-05-15:
cool, did not noticed that since it was closed and open again. but good hear cylon is working great for you. Let us know if you have any more issues or notice something that can be improved, thanks.

deadprogram, 2015-05-15:
It is one of the more common problems that people run into, so we're adding some code to detect that. @ddm you are helping us by finding things that are not intuitive, and we thank you!

Kumaresh-Rajalingam, 2017-04-20:
While testing leapmotion with arduino it showing **optimized for desktop usage** how to solve this error Help