hybridgroup/cylon #302: Error while running The "Hello, World" Of Things example

ShashankaNataraj, 2015-06-25:
Hi,

I have the following error while running the following error while running the hello world example:

[image]

Im on elementaryos which is based on ubuntu.

I executed: 
`sudo apt-get update && sudo apt-get install arduino arduino-core`  as mentioned in [this article](http://playground.arduino.cc/Linux/Ubuntu)

Then 
`npm install cylon-firmata cylon-gpio cylon-i2c` as mentioned in the cylon website. Is this a known issue?

ShashankaNataraj, 2015-06-25:
Ok I just sudo'd `sudo node helloworld.js` and can now see the LED blink when the program gets uploaded to the arduino, however, the program gets stuck at starting connections?

edgarsilva, 2015-06-25:
hi @ShashankaNataraj,

Did you uploaded the firmata sketch to the arduino?

ShashankaNataraj, 2015-06-25:
Hi, how do I upload the firmata sketch? Nothing is mentioned in the hello world example.

Im using an UNO, yes.

edgarsilva, 2015-06-25:
Here: https://github.com/hybridgroup/cylon-firmata#ubuntu

You can use [gort.io](http://gort.io/) or upload directly using the arduino IDE software, just open it go to file > examples > firmata > StandardFirmata.ino open it and upload to the Arduino Uno.

edgarsilva, 2015-07-09:
@ShashankaNataraj were you abel to upload firmata and run the examples?

ShashankaNataraj, 2015-07-09:
@edgarsilva Not yet, got sidetracked with work :( Will check again and let you know. Last I remember I couldn't get firmata uploaded with this yet.

edgarsilva, 2015-07-09:
@ShashankaNataraj ok, let me know, you can always just use the Arduino IDE to upload firmata, go to File > Examples > Firmata > StandardFirmata and upload to the arduino.

ShashankaNataraj, 2015-07-10:
@edgarsilva Actually that I was able to do yesterday. Now I just have to run cylonJS and try if it works. WIll report back today evening.

deadprogram, 2015-09-01:
Hi, @ShashankaNataraj were you able to get your example working?

ShashankaNataraj, 2015-09-01:
Hey @deadprogram Thanks for asking, but got stuck with some other work. Ill close this issue as of now...

closed 2015-09-01 by ShashankaNataraj