hybridgroup/cylon #307: ESC control, no way to go backwards.

r2DoesInc, 2015-08-05:
I have my brushless motors working with the ESC using your servo library, where the "angle" I provide is actually the speed the motors should run. This is the behavior I would expect coming from other libraries that did the same.

However, with cylon, I cannot seem to get the motor to spin backwards. Setting a negative number, or anything below 90 just stops the wheels. 

With 90 as neutral, I would expect 0-89 as backwards movement, and 91-180 as forward movement.

edgarsilva, 2015-08-05:
@r2DoesInc I'm not sure that is the driver you should be using. Do these brushless motors work as regular servos? Can you pls provide the model so I can check the data sheet for inputs and how they work?

r2DoesInc, 2015-08-05:
I found on the arduino driven version of the project, that the ESC basically gets treated as a servo. I had that version working perfect, forward and backward. 

The car im attempting to drive is the duratrax evader brushless.
http://manuals.hobbico.com/dtx/dtxd37-manual.pdf
Unfortunately I dont see an ESC model number, just a few specs listed, 
`Voltage Input:
BEC Max:
Current Rating:
Motor Limit:
2S LiPo, 6-7 cell NiCd or NiMH
3 Amps @ 2S input
Up to 100A
On 2S: 4200kV Max`

edgarsilva, 2015-08-05:
mmm.. maybe the period in the servo library might be incorrect for the motors, 50hz rating is too slow. You can also try the the motor driver, the `speed` method works as PWM at a higher freq, around 2000 by default I think. I'm trying to find a motor to test this myself, I might have one shortly.

r2DoesInc, 2015-08-05:
I attempted to use the motor driver to no avail. I also thought about trying to continious servo driver as it has both directions, but I couldnt think of a way to get variable speed there so I didnt make that attempt.

edgarsilva, 2015-08-05:
@r2DoesInc Can you pls share the arduino code that is working?

r2DoesInc, 2015-08-05:
https://github.com/cloudspace/ArdrobotAndroid/blob/master/Resources/Arduino/motor_controller/motor_controller.ino

edgarsilva, 2015-08-05:
What are you running this on? Arduino? Raspi? BBB? edison?

r2DoesInc, 2015-08-05:
That was for the MegaADK

edgarsilva, 2015-08-05:
There used to be some issues with Firmata and the mega before, not sure if they have been fixed. A couple of things, you are only connecting the `signal` cable and `gnd` from the ESC to the arduino right? the power is coming from an external source? do you have any servo to test the driver with that board? besides the brushless motors that is

r2DoesInc, 2015-08-05:
The current project is Edison based.
I have continuous and standard servos working as expected.
I have pwr/sig/gnd from esc to the edison. The power is coming from the large lipo in the car.

I have also powered it externally through the micro usb port without connecting the pwr from esc to edison.

edgarsilva, 2015-08-07:
Tested the servos as you mentioned they work just fine. Does the ESC have a model number or datasheet or something we can check? maybe is an issue with the period/duty/freq or the pulse width.

All of this can be configure in the servo driver, but we do need a datasheet to corroborate the values.

r2DoesInc, 2015-08-10:
The ESC on the Evader is pretty much a rebranded version of [this](http://www.castlecreations.com/products/sidewinder.html). I know that it supports reverse, and that reverse can be disabled. I can also confirm that reverse is not currently disabled, as I can go in reverse if i reconnect the original radio controller, or if i hook it back up to the original arduino project.

edgarsilva, 2015-08-10:
@r2DoesInc could you pls share the cylon (JS) code you are using?

r2DoesInc, 2015-08-10:
https://github.com/cloudspace/ArdrobotEdison/blob/master/server.js

Ive also tried not using the safeAngle option, and wrote a quick test to send it -180 to 180 values. As soon as it ticked over 90, the motors began to spin forward.

edgarsilva, 2015-08-10:
The servo driver always uses `safeAngle()` even when calling `angle()`, you can override this by passing ` range: { max: 180, min: 0 }` as follows:

```javascript
devices: {
  turnServo: {
    driver: 'servo',
    pin: 3,
    range: { min: 0, max 180 }
  },
  driveServo: {
    driver: 'servo',
    pin: 5,
    range: { min: 0, max 180 }
  }
},
```

This part in your code is also not necessary: https://github.com/cloudspace/ArdrobotEdison/blob/master/server.js#L40-L50

it should look something like this:

```javascript
          setAngle: function(val) {
            if (val != null) {
                this.turnServo.angle(val);
            }
        },

        setSpeed: function(val) {
            if (val != null) {
                this.driveServo.angle(val);
            }
        },
```

You don't need to use `safeAngle()`, it is used internally by `angle()`, and can be overridden as we did above. not sure why the `+val`  though.

Could you please give this a try and let me know how it goes?

r2DoesInc, 2015-08-12:
The `+val` is just from my unfamiliarity with typeless languages coming from Java, you can disregard that.

Ive set the range in the device info as you mentioned, and did away with the `safeAngle` code. Unfortunately the motors are still not running backwards, anything under 90 just brings them to a halt.

Its interesting to note though, that the esc requires a full reverse signal to arm as you can see  [here] (https://github.com/cloudspace/ArdrobotEdison/blob/master/server.js#L63). The arming sequence works, so that full reverse value is being written correctly - as you can also confirm by writing an angle <80 and then querying the `currentAngle`.

deadprogram, 2015-11-08:
Hi, @r2DoesInc & @edgarsilva 

Is this issue considered resolved? Also, probably worth pointing out that this conversation should probably be on the https://github.com/hybridgroup/cylon-gpio repo :)

r2DoesInc, 2015-11-09:
The company I was working for on this project has since let me go, so I
have not kept up to date on their goings on. I dont imagine they are
pursuing the project any further.

deadprogram, 2015-11-09:
Thanks for the update. Closing.

closed 2015-11-09 by deadprogram