hybridgroup/cylon #310: Windows 10 Connection

vincent-leonardo, 2015-08-20:
I'm currently on Windows 10.
Attempting to connect a Sphero but it always returns:
```
events.js:85
    throw er; // Unhandled 'error' event

Error: Opening \\.\COM*: Unknown error code 121
    at Error (native)
```
and in some case the error code is 1168.

I tried to connect to the Sphero using https://github.com/eelcocramer/node-bluetooth-serial-port, it could connect to it, but it uses MacAddress with Channel instead of COM port. Is there a solution to it? Or anyway i could setup the connetion through node-bluetooth-serial-port and utilize the functions of Cylon.

edgarsilva, 2015-08-20:
Hi @vincent-leonardo ,

I don't have a windows 10 platform to test yet, it is probable that the Bluetooth dev stack changed and the connections or libraries we use for BT do not work there yet.

This looks like it is an issue with the serialport connection, you can test if node-serialport works by following the steps in their readme: https://github.com/voodootikigod/node-serialport

If serial port does not work then cylon interfaces that use it won't work, if it does work then we can follow the rabbit hole and fix it.

I will take a look when I have a chance, thanks for the heads up.

Edgar Silva

vincent-leonardo, 2015-08-21:
I forgot to mention it, I did test it out on node-serialport as well after I realize that cylon uses it as its dependency, I got the similar error message.

Thanks a lot, in the meantime I'll also try to find the possible solution.

vincent-leonardo, 2015-08-21:
I'm closing in to the solution. I've created an adaptor directly in the Sphero module with reference to its original serialport adaptor. This adaptor works as per normal adaptor except it handles bluetooth device's MacAddress and automatically find it's service RFCOMM port.

```javascript
// adaptor/bluetooth-serialport.js
"use strict";

var util = require("util"),
    EventEmitter = require("events").EventEmitter;

var serialport,
    isChrome = typeof chrome !== "undefined";

try {
  serialport = require("bluetooth-serial-port");
} catch (error) {
  serialport = null;
}

if (serialport == null) {
  var err = [
    "It looks like bluetooth-serial-port didn't compile properly.",
    "This is a common problem, and it's fix is documented here:",
    "https://github.com/eelcocramer/node-bluetooth-serial-port"
  ].join(" ");

  console.error(err);
  throw new Error("Missing bluetooth-serial-port dependency");
}

/**
 * An adaptor to communicate with a serial port
 *
 * @constructor
 * @param {String} conn the serialport string to connect to
 */
var Adaptor = module.exports = function Adaptor(conn) {
  this.conn = conn;
  this.serialport = null;
};

util.inherits(Adaptor, EventEmitter);

/**
 * Opens a connection to the serial port.
 * Triggers the provided callback when ready.
 *
 * @param {Function} callback (err)
 * @return {void}
 */
Adaptor.prototype.open = function open(callback) {
  var self = this,
      port = this.serialport = new serialport.BluetoothSerialPort();
  port.findSerialPortChannel(self.conn, function(channel) {
    console.log("Opening: " + self.conn + " on " + channel);
    port.connect(self.conn, channel, function(){
      self.emit("open");
      port.on("error", emit("error"));
      port.on("close", emit("close"));
      port.on("data", emit("data"));
      callback();
    });
  }, function(err) {
      console.log('Error searching ' + self.conn + ': ' + err);
  });
  
  function emit(name) {
    return self.emit.bind(self, name);
  }
};

/**
 * Writes data to the serialport.
 * Triggers the provided callback when done.
 *
 * @param {Any} data info to be written to the serialport. turned into a buffer.
 * @param {Function} [callback] triggered when write is complete
 * @return {void}
 */
Adaptor.prototype.write = function write(data, callback) {
  this.serialport.write(data, callback);
};

/**
 * Adds a listener to the serialport's "data" event.
 * The provided callback will be triggered whenever the serialport reads data
 *
 * @param {Function} callback function to be invoked when data is read
 * @return {void}
 */
Adaptor.prototype.onRead = function onRead(callback) {
  this.on("data", callback);
};

/**
 * Disconnects from the serialport
 * The provided callback will be triggered after disconnecting
 *
 * @param {Function} callback function to be invoked when disconnected
 * @return {void}
 */
Adaptor.prototype.close = function close(callback) {
  this.serialport.close(callback);
};
```

And I also modified its loader to handle MacAddress in the format of (XX:XX:XX:XX).

```javascript
// loader.js
"use strict";

function isSerialPort(str) {
  // use regexp to determine whether or not 'str' is a serial port
  return /(\/dev\/|com\d+).*/i.test(str);
}

function isBluetoothPort(str) {
  return /\(?([0-9A-F]{2}[:-]){5}([0-9A-F]{2})\)?/i.test(str);
}

/**
 * Loads an adaptor based on provided connection string and system state
 *
 * @param {String} conn connection string (serial port or BLE UUID)
 * @return {Object} adaptor instance
 */
module.exports.load = function load(conn) {
  var isSerial = isSerialPort(conn),
      isBluetooth = isBluetoothPort(conn),
      isChrome = typeof chrome !== "undefined",
      Adaptor;

  if (isSerial) {
    Adaptor = require("./adaptors/serialport");
  } else if (isBluetooth){
    Adaptor = require("./adaptors/bluetooth-serialport")
  } else if (isChrome) {
    // load chrome BLE adaptor
  } else {
    // load BLE adaptor (noble?)
  }

  return new Adaptor(conn);
};
```

I could get it to connect to my sphero now. But I could not execute any command both natively using Sphero.js or through Cylon.js.
I'm always getting this error message:
```
Assertion failed: obj->HasIndexedPropertiesInExternalArrayData(), file src\node_buffer.cc, line 103
```
Any idea of which part I might have left out?

closed 2015-08-24 by vincent-leonardo