hybridgroup/cylon #313: list of user devices

vkfont, 2015-10-19:
so, i've created UPC monitoring robot (can send alert if power is low or electricity is disabled, useful for home automation and other purposes)
now, how can i publish it so anyone who use cylon.js can see and use it? is there any naming conventions for npm module? or perhaps you want to create actual catalogue on site?

deadprogram, 2015-10-19:
Hi @vkfont that sounds great!

One common pattern for "User Created Platforms" is to publish yourself via `npm`. We suggest naming it `cylon-X` for example, possibly `cylon-upc` in your case.

Once available, please submit a PR against the README in this site:
https://github.com/hybridgroup/cylon/blob/master/README.markdown#hardware-support

Scroll down a little ways, and you will see "User Created Platforms"
 
and also our documentation site:
https://github.com/hybridgroup/cylon-site/blob/master/source/index.html.haml#L108

Thank you!

afoninsky, 2015-10-19:
my fault, thank you :)

closed 2015-10-19 by vkfont