hybridgroup/cylon #314: [Docs] more examples using testing

7yl4r, 2015-11-03:
I've been trying to set up a basic robot along with a test, but am having a bit of trouble I'm unable to figure out. A few more examples which use testing would definitely help. 

Here's a gist with [my minimal robot.js and robot_test.js](https://gist.github.com/7yl4r/744db0ebe11252144aa2) which is a stripped-down hello-world example. When I run the tests I get

```
2015-11-03T14:29:46.984Z : [cylon-test-bot] - Starting connection 'loopback'.
/home/user/cylon-test-bot/node_modules/cylon/lib/adaptor.js:60
  throw new Error(message);
  ^

Error: Error in connection 'loopback' - Adaptor#connect method must be overwritten by descendant classes.
    at TestAdaptor.Adaptor.connect (/home/user/cylon-test-bot/node_modules/cylon/lib/adaptor.js:60:9)
    at Robot.startConnection (/home/user/cylon-test-bot/node_modules/cylon/lib/robot.js:317:22)
    at Robot.<anonymous> (/home/user/cylon-test-bot/node_modules/cylon/lib/robot.js:288:19)
    at /home/user/cylon-test-bot/node_modules/cylon/lib/utils/helpers.js:237:36
    ...
```

I haven't quite been able to make sense of this, but I'm guessing I need to mock the connection?

I've also tried removing the `loopback

deadprogram, 2015-11-08:
Hi, @7yl4r 

I made some corrections to the test drive robotics example in the docs site here: https://github.com/hybridgroup/cylon-site/commit/3f6dc7f16c7b6d28378710ffbb8dad3796fa6327

Thanks for pointing out that the example was not correct! It needed to be updated for a while.

Test Driven Robotics (TDR) is a subject that probably deserves an entire blog post/book on its own. However, please let me know if the above is sufficient to unblock you and close this particular issue.

llwoll, 2016-01-01:
Hello,  one error  happened when I update from 1.0.0 to 1.2.0
I am sure it is well when i use 1.0.0

Following is the error

       > TypeError: Cannot read property 'IOT' of undefined

Following is my source code 

      Cylon.robot({
              name: 'IOT',
             description: 'Description is optional...'
              .......
      });

the error happened when i call  that

         Cylon.robots['IOT']        

SORRY FOR MY POOR ENGLISH,TANKS

deadprogram, 2016-01-09:
Hi, @llwoll the correct syntax is now:

```
Cylon.MCP.robots["IOT"];
```

The documentation was updated as part of https://github.com/hybridgroup/cylon-site/commit/3f6dc7f16c7b6d28378710ffbb8dad3796fa6327

Hope that helps!

deadprogram, 2016-01-19:
Since this issue appears to be resolved, I am going to close it. Please reopen if needed.

closed 2016-01-19 by deadprogram

geryb-bg, 2016-07-17:
Hi

I am still seeing this same issue. I copied the example exactly as it appears on the website but am getting the same error that @7yl4r described. Any chance you could help me?

Thanks