hybridgroup/cylon #315: Cylon require fails using webpack

MrSpijker, 2015-11-03:
Hi, 

I'm using webpack to bundle my (node) scripts but when I require Cylon I get the following errors:
[image]

I have a workaround by adding:

node: {
  fs: 'empty',
  readline: 'empty'
}

in my webpack config file but then I get the following: 
[image]

cylon-firmata is installed under my node_modules.
I don't have any problems with other modules -_-

deadprogram, 2015-11-03:
I've not yet done anything with webpack, so far only with browserfy.

https://github.com/hybridgroup/cylon-example-browserify

Seems like it might be related to how the requires are handled differently? I found a few articles on a web search about this, but did not have a chance to really dig in yet.

MrSpijker, 2015-11-03:
We started building everything in a webpack ecosystem so going back to browserify is not a preferred option. Thanks for letting me now it does work in a browserified environment.

deadprogram, 2015-11-03:
Well, some cylon modules do. We have more info about all that here: http://cylonjs.com/documentation/guides/browser-support/

MrSpijker, 2015-11-03:
chrome does not make the difference. Was using safari earlier. -_- 
what if i'd communicate with a non browser module, only build to control the arduino's using, by example, spacebrew? 
It's an option but i'd prefer to keep it in my web bundle

deadprogram, 2015-11-04:
You should be able to do a build of your app, cylon, cylon-firmata, & firmata module that use browser-serialport, but not sure how to do that with webpack.