hybridgroup/cylon #316: Can't install the "firmata" package

jpjamipark, 2015-11-10:
Hello, I'm having trouble installing the firmata package. Below is my output. I've looked at a previous similar thread, which suggested going to a prev version, so I tried installing node 4.2.2 and trying that. Similar results occurred.
```
> serialport@2.0.4 install /usr/local/lib/node_modules/cylon-firmata/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: *** No rule to make target `../../../../../../../../Volumes/Storage', needed by `Makefile'.  Stop.
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 14.5.0
gyp ERR! command "/usr/local/Cellar/node/5.0.0/bin/node" "/usr/local/lib/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/local/lib/node_modules/cylon-firmata/node_modules/serialport/build/Release/node-v47-darwin-x64/serialport.node" "--module_name=serialport" "--module_path=/usr/local/lib/node_modules/cylon-firmata/node_modules/serialport/build/Release/node-v47-darwin-x64"
gyp ERR! cwd /usr/local/lib/node_modules/cylon-firmata/node_modules/serialport
gyp ERR! node -v v5.0.0
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/Cellar/node/5.0.0/bin/node /usr/local/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/cylon-firmata/node_modules/serialport/build/Release/node-v47-darwin-x64/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/cylon-firmata/node_modules/serialport/build/Release/node-v47-darwin-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/cylon-firmata/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:818:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Darwin 14.5.0
node-pre-gyp ERR! command "/usr/local/Cellar/node/5.0.0/bin/node" "/usr/local/lib/node_modules/cylon-firmata/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/local/lib/node_modules/cylon-firmata/node_modules/serialport
node-pre-gyp ERR! node -v v5.0.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.13
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/Cellar/node/5.0.0/bin/node /usr/local/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/cylon-firmata/node_modules/serialport/build/Release/node-v47-darwin-x64/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/cylon-firmata/node_modules/serialport/build/Release/node-v47-darwin-x64' (1)
npm ERR! Darwin 14.5.0
npm ERR! argv "/usr/local/Cellar/node/5.0.0/bin/node" "/usr/local/bin/npm" "install" "-g" "cylon-firmata"
npm ERR! node v5.0.0
npm ERR! npm  v3.3.6
npm ERR! code ELIFECYCLE

npm ERR! serialport@2.0.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@2.0.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Volumes/Storage Macbook/JPark/Documents/Development/LeapRPS/npm-debug.log
```

deadprogram, 2015-11-10:
Hi @jpjamipark 

It appears that the node-serialport npm module was unable to install/build on your Mac. Your machine's configuration does not match the criteria for any of the precompiled binaries for the latest node-serialport build that @jacobrosenthal just releases, so node-pre-gyp is doing a `--fallback-to-build` meaning build from sources on your machine.

You might want to try to follow this advice: https://github.com/meanjs/mean/issues/946#issuecomment-144362004 to install the OSX build tools.

jacobrosenthal, 2015-11-10:
> You might want to try to follow this advice: meanjs/mean#946 (comment)

What version serialport?

deadprogram, 2015-11-10:
@jpjamipark is trying to install the latest `2.0.4`

jacobrosenthal, 2015-11-10:
Ah, node 47, ie node 5, Yeah we dont nor have ever had have binaries for node5. Check out the building from source dependencies https://github.com/voodootikigod/node-serialport#installation-special-cases

deadprogram, 2015-11-10:
@jpjamipark does that get you going in the right direction? 

Also, side note: this issue probably belongs in the https://github.com/hybridgroup/cylon-firmata repo, rather than the main Cylon.js repo.

jpjamipark, 2015-11-10:
I downgraded to node 0.12.7, and got similar results. I also verified that xCode command line tools were installed. Same results. I also previously downgraded to 4.2.2 when trying to install, with similar results. Sorry about posting in the cylon repo, I wasn't aware of the cylon firmata repo.

deadprogram, 2015-11-10:
Can you provide the logs for all your failed output? 

The easiest is a node version that matches one of the available precompiled node-serialport binaries: https://github.com/voodootikigod/node-serialport/releases/tag/v2.0.4

jpjamipark, 2015-11-10:
Here are the results:
```
> serialport@2.0.4 install /usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: *** No rule to make target `Macbook/JPark/.node-gyp/0.12.7/common.gypi', needed by `Makefile'.  Stop.
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.5.0
gyp ERR! command "/usr/local/Cellar/node/5.0.0/bin/node" "/usr/local/lib/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport/build/Release/node-v14-darwin-x64/serialport.node" "--module_name=serialport" "--module_path=/usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport/build/Release/node-v14-darwin-x64"
gyp ERR! cwd /usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/Cellar/node/5.0.0/bin/node /usr/local/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport/build/Release/node-v14-darwin-x64/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport/build/Release/node-v14-darwin-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:1015:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1087:5)
node-pre-gyp ERR! System Darwin 14.5.0
node-pre-gyp ERR! command "node" "/usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport
node-pre-gyp ERR! node -v v0.12.7
node-pre-gyp ERR! node-pre-gyp -v v0.6.13
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/Cellar/node/5.0.0/bin/node /usr/local/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport/build/Release/node-v14-darwin-x64/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/cylon-firmata/node_modules/firmata/node_modules/serialport/build/Release/node-v14-darwin-x64' (1)
npm ERR! Darwin 14.5.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "cylon-firmata"
npm ERR! node v0.12.7
npm ERR! npm  v2.11.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@2.0.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@2.0.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Volumes/Storage Macbook/JPark/Documents/Development/LeapRPS/npm-debug.log
```

deadprogram, 2015-11-10:
@jacobrosenthal anything in that build failure that looks familiar to you? He is using Node `v0.12.71

deadprogram, 2015-11-17:
Hi, @jpjamipark 

Since this is actually related to node-serialport and your machine's OSX configuration, I would recommend you create an issue on the https://github.com/voodootikigod/node-serialport repo to get it resolved.

I am closing this issue. Please reopen if I can provide further info.

closed 2015-11-17 by deadprogram