hybridgroup/cylon #317: Can I use cylon with Airborne Cargo Drones?

AlbertoHdezCerezo, 2015-11-10:
Hello everyone,

Right now I am planning to develop for fun a project based on Meteor JS which involves drones. Checking your documentation I saw that right now there is not support for Parrot Airborne Cargo Drones (right now it is the drone I am considering), however for what I see in some of your blog posts, sometimes you suggest using nodecopter.

My doubt it is first cylon or nodecopter.js is compatible with Airborne Cargo Drones, and second, if there is any plan to release an updated version of the project with support for these devices.

That's all, thank you so much for your help and this incredible project!

deadprogram, 2015-11-10:
Hi @AlbertoHdezCerezo 

The Parrot Airborne Cargo Drone is a variation of the Parrot Rolling Spider. There is Cylon.js module for the Rolling Spider called https://github.com/ChrisTheBaron/cylon-rolling-spider created by @ChrisTheBaron that should work for what you need.

So the answer should be "yes".

AlbertoHdezCerezo, 2015-11-10:
Nice, thank you very much for your help!

closed 2015-11-10 by AlbertoHdezCerezo