hybridgroup/cylon #318: Cannot find the 'cylon-undefined' module.

kenmarold, 2015-11-13:
I'm trying to run a basic led script on arduino and I'm getting the error: 
    Cannot find the 'cylon-undefined' module.
    This problem might be fixed by installing it with 'npm install cylon-undefined' and trying again.
    Terminated: 15

When I run npm install cylon-undefined in Terminal I get the error: 

npm ERR! Darwin 15.0.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "cylon-undefined"
npm ERR! node v4.2.2
npm ERR! npm  v2.14.7
npm ERR! code E404

npm ERR! 404 Registry returned 404 for GET on https://registry.npmjs.org/cylon-undefined
npm ERR! 404 
npm ERR! 404 'cylon-undefined' is not in the npm registry.
npm ERR! 404 You should bug the author to publish it (or use the name yourself!)
npm ERR! 404 
npm ERR! 404 Note that you can also install from a
npm ERR! 404 tarball, folder, http url, or git url.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/kenmarold/Arduino/CylonJS/npm-debug.log

Can someone tell me what I'm doing wrong here? Thanks!

Debug file attached below:
[npm-debug.pdf](https://github.com/hybridgroup/cylon/files/34086/npm-debug.pdf)

deadprogram, 2015-11-13:
Hi @kenmarold 

If you want to use an Arduino you need to use `cylon-firmata` & friends. To install, you need to run:

`npm install cylon-firmata cylon-gpio cylon-i2c`

The docs page for using Cylon.js with Arduino might be useful as well: http://cylonjs.com/documentation/platforms/arduino/

Hope that helps!

kenmarold, 2015-11-13:
Thanks! I'll re-check everything.

deadprogram, 2015-12-05:
Hi @kenmarold were you able to resolve your issue?

Japh, 2016-01-08:
I'm having this same issue trying to use `cylon-spark`.

```
Cannot find the 'cylon-undefined' module.
This problem might be fixed by installing it with 'npm install cylon-undefined' and trying again.
Terminated
```

I'm assuming this basically means that for some reason, `spark` isn't being set correctly for the autoloading of the `cylon-spark` module by `cylon`.

My code is exactly the LED example code from https://github.com/hybridgroup/cylon-spark .

deadprogram, 2016-01-09:
@Japh did you run `npm install cylon cylon-spark cylon-gpio` or something like that to install the dependencies before running your code?

Japh, 2016-01-11:
@deadprogram I sure did :smile:

deadprogram, 2016-01-11:
Also, note that you should not install these modules globally, just in the project itself. 

I just tried the same code as in the repo on a fresh install using Node 4.2.2 with a local install, and did not have any such problem. Which OS and Node version are you using?

Japh, 2016-01-11:
Ah-ha! I'd rolled back to an older version of Node (separate project requirement) and forgotten to update again.

Thanks! Working now  :smile:

iyogeshjoshi, 2016-02-05:
I'm trying to run the example `cylon-pebble` code but getting same error "Cannot find the `cylon-undefined` module." .

I tried installing all modules mentioned in earlier comments but nothing helped.

Note: I'm running it on Mac with _OS X El Capitan_ and node version `v4.2.6`

deadprogram, 2016-02-06:
I might know what's happening here. @iyogeshjoshi did you git clone the `cylon-pebble` repo, and then tried to run those examples from within the cloned directory?

iyogeshjoshi, 2016-02-06:
@deadprogram No no I wrote the example, and did the `npm install` for required modules.

deadprogram, 2016-02-06:
OK, let me try a few things, and I see if I can repro.

iyogeshjoshi, 2016-02-10:
@deadprogram found anything?

deadprogram, 2016-02-10:
Hi, @iyogeshjoshi which exact example are you trying?

iyogeshjoshi, 2016-02-10:
@deadprogram  [here](https://gist.github.com/iyogeshjoshi/38ce663e25f7b8b6ac1a.js) is my code for example and package.json file.

deadprogram, 2016-02-10:
In your example code, https://gist.github.com/iyogeshjoshi/38ce663e25f7b8b6ac1a#file-cylon-example-js-L13

should be "adaptor" not "adapter".

Probably we should allow either spelling.

iyogeshjoshi, 2016-02-16:
@deadprogram :+1:  yeah that worked, and yeah adding both spelling is a good idea I guess. 
Guide me I would like to do it :)

deadprogram, 2016-02-16:
Hi, @iyogeshjoshi the code that constructs a connection is here: https://github.com/hybridgroup/cylon/blob/master/lib/robot.js#L109 which would probably be a good place to handle it.

deadprogram, 2016-02-25:
This has now been corrected by https://github.com/hybridgroup/cylon/pull/331

closed 2016-02-25 by deadprogram