hybridgroup/cylon #321: errors and warnings installing cylon-sphero modules

TimmyV, 2015-11-19:
ran this in Windows command prompt:

 npm install cylon cylon-sphero

got a whole bunch of errors(see attachment) Can't figure out how to get cylon-sphero libraries installed.

Thanks for your help.

[errmsg.txt](https://github.com/hybridgroup/cylon/files/38479/errmsg.txt)

deadprogram, 2015-11-19:
Hi, @TimmyV 

This is due to https://github.com/orbotix/sphero.js/issues/31 which will hopefully be fixed very soon.

deadprogram, 2015-12-01:
Hi, @TimmyV 

We released a new version of cylon-sphero last week that should correct your problem:

https://github.com/hybridgroup/cylon-sphero/releases/tag/v0.23.0

Sorry, forgot to update this issue. Please give it a try!

deadprogram, 2015-12-03:
I am going to close this issue, please reopen if you require further assistance.

closed 2015-12-03 by deadprogram