hybridgroup/cylon #322: Cannot find the 'cylon-firmata' module.

manu354, 2015-12-03:
I did all the steps in http://cylonjs.com/documentation/platforms/arduino/ for windows and then created the script.js file with this: 

`var Cylon = require("cylon");

// Initialize the robot
Cylon.robot({
  // Change the port to the correct port for your Arduino.
  connections: {
    arduino: { adaptor: 'firmata', port: 'COM3' }
  },

  devices: {
    led: { driver: 'led', pin: 13 }
  },

  work: function(my) {
    every((1).second(), function() {
      my.led.toggle();
    });
  }
}).start()`

I then installed the packagess:

`$ npm install cylon-firmata cylon-gpio cylon-i2c`

But when I try to run the script with node I get this error:

`$ c:\web>node script`

`Cannot find the 'cylon-firmata' module.
This problem might be fixed by installing it with 'npm install cylon-firmata' and trying again.`

I tried reinstalling cylon-firmata but I get the same error.

deadprogram, 2015-12-03:
Hi, @manu354 

We will try to reproduce your issue. Can you please let us know the version of Windows and Node.js that you are using?

deadprogram, 2015-12-03:
Also, can you please provide a directory listing of your `node_modules` directory? Thanks.

manu354, 2015-12-03:
I downloaded the latest node to see if that would fix the problem. Windows version is Microsoft Windows 10 Home 10.0.10240 build 10240. 

node_modules has 

`.bin browser-serialport cylon cylon-gpio cylon-i2c es6shmin`

I might be getting an error while installing cylon. I can't tell. Should i paste the install messages?

deadprogram, 2015-12-03:
Interesting. Yes, please.

manu354, 2015-12-03:
`c:\web>npm install cylon-firmata cylon-gpio cylon-i2c

> serialport@2.0.5 install c:\web\node_modules\serialport
> node-pre-gyp install --fallback-to-build

c:\web\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" clean )  else (node  clean )

c:\web\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" configure --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64 )  else (node  configure --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64 )
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:116:14)
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:71:11
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:82:15)
gyp ERR! System Windows_NT 10.0.10240
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "--fallback-to-build" "--module=c:\\web\\node_modules\\serialport\\build\\Release\\node-v47-win32-x64\\serialport.node" "--module_name=serialport" "--module_path=c:\\web\\node_modules\\serialport\\build\\Release\\node-v47-win32-x64"
gyp ERR! cwd c:\web\node_modules\serialport
gyp ERR! node -v v5.1.0
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd configure --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (c:\web\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:818:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Windows_NT 10.0.10240
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "c:\\web\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd c:\web\node_modules\serialport
node-pre-gyp ERR! node -v v5.1.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.13
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd configure --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64' (1)
npm WARN install:serialport@2.0.5 serialport@2.0.5 install: `node-pre-gyp install --fallback-to-build`
npm WARN install:serialport@2.0.5 Exit status 1
c:\web
├── browser-serialport@2.0.3
├─┬ cylon-gpio@0.27.0
│ └── cylon@1.2.0
├── cylon-i2c@0.23.0
└── es6-shim@0.33.13

npm WARN ENOENT ENOENT: no such file or directory, open 'c:\web\package.json'
npm WARN EPACKAGEJSON web No description
npm WARN EPACKAGEJSON web No repository field.
npm WARN EPACKAGEJSON web No README data
npm WARN EPACKAGEJSON web No license field.
npm WARN EBUNDLEOVERRIDE Replacing bundled node_modules\serialport\node_modules\node-pre-gyp-github with new installed version

c:\web>
c:\web>> serialport@2.0.5 install c:\web\node_modules\serialport

c:\web>> node-pre-gyp install --fallback-to-build`

manu354, 2015-12-04:
I tried installing cylon cylon-firmata and the cylon-firmata deletes itself at the end of the npm install. The same thing happens if i just install cylon-firmata. I tried copying the module before it deleted itself and pasting it back in but i get the same error. Maybe I can  download the module somewhere?

deadprogram, 2015-12-04:
Hi, @manu354 

Looks like the problem here is that `node-serialport` which is one of `cylon-firmata` module dependencies was not able to install properly.

There should be precompiled binaries for node-serialport for versions Node.js 0.10, 0.12, & 4.2 for Windows already, if you were to install under one of those versions. Otherwise you will need to install the prerequisites as detailed here: https://github.com/voodootikigod/node-serialport#windows

Hope that helps!

manu354, 2015-12-04:
Same thing happens after downloading python. 
c:\web>npm install cylon cylon-firmata

> serialport@2.0.5 install c:\web\node_modules\serialport
> node-pre-gyp install --fallback-to-build

c:\web\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" clean )  else (node  clean )

c:\web\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" configure --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64 )  else (node  configure --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64 )

c:\web\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" build --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64 )  else (node  build --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64 )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
MSBUILD : error MSB4132: The tools version "2.0" is unrecognized. Available tools versions are "4.0".
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Windows_NT 10.0.10240
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=c:\\web\\node_modules\\serialport\\build\\Release\\node-v47-win32-x64\\serialport.node" "--module_name=serialport" "--module_path=c:\\web\\node_modules\\serialport\\build\\Release\\node-v47-win32-x64"
gyp ERR! cwd c:\web\node_modules\serialport
gyp ERR! node -v v5.1.0
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd build --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (c:\web\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:818:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Windows_NT 10.0.10240
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "c:\\web\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd c:\web\node_modules\serialport
node-pre-gyp ERR! node -v v5.1.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.13
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd build --fallback-to-build --module=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node --module_name=serialport --module_path=c:\web\node_modules\serialport\build\Release\node-v47-win32-x64' (1)
npm WARN install:serialport@2.0.5 serialport@2.0.5 install: `node-pre-gyp install --fallback-to-build`
npm WARN install:serialport@2.0.5 Exit status 1
c:\web
├── browser-serialport@2.0.3
├── cylon@1.2.0
├── cylon-gpio@0.27.0
├── cylon-i2c@0.23.0
└── es6-shim@0.33.13

npm WARN ENOENT ENOENT: no such file or directory, open 'c:\web\package.json'
npm WARN EPACKAGEJSON web No description
npm WARN EPACKAGEJSON web No repository field.
npm WARN EPACKAGEJSON web No README data
npm WARN EPACKAGEJSON web No license field.
npm WARN EBUNDLEOVERRIDE Replacing bundled node_modules\serialport\node_modules\node-pre-gyp-github with new installed version

c:\web>node script
Cannot find the 'cylon-firmata' module.
This problem might be fixed by installing it with 'npm install cylon-firmata' and trying again.

deadprogram, 2015-12-04:
Hi, @manu354 

From your output from the failed `node-serialport` build, I see this:

```
MSBUILD : error MSB4132: The tools version "2.0" is unrecognized. Available tools versions are "4.0".
```

I did a web search, and appears that you might want to try the suggestion here: http://stackoverflow.com/questions/14278417/cannot-install-node-modules-that-require-compilation-on-windows-7-x64-vs2012/22411007#22411007

deadprogram, 2015-12-04:
Worth mentioning here, is that you should be able to use the precompiled binary of `node-serialport` if you downgrade to Node.js version 4.2, which will help you avoid needing to install the whole compilation toolchain on Windows. I have submitted a PR to `node-serialport` that will provide a precompiled binary for Node 5.1 which will be especially useful for Windows users:

https://github.com/voodootikigod/node-serialport/pull/640

Hopefully it will get merged and a new release put out soon.

manu354, 2015-12-05:
Where can I find a precompiled binary?

deadprogram, 2015-12-05:
If you install under Node.js 4.2 in Windows, it will automatically install the precompiled binary of `node-serialport` for you. 

Here is some info about how it works: https://github.com/voodootikigod/node-serialport/blob/master/publish-binaries.md

manu354, 2015-12-06:
Still doesn't work after doing http://stackoverflow.com/questions/14278417/cannot-install-node-modules-that-require-compilation-on-windows-7-x64-vs2012/22411007#22411007 Ill just downgrade to 4.2.2

manu354, 2015-12-06:
Downgrading to 4.2.2 fixes it.  Thanks for all the help!!!

deadprogram, 2015-12-06:
Glad that helped... closing.

closed 2015-12-06 by deadprogram