hybridgroup/cylon #323: Using yun shield with cylonjs.

manu354, 2015-12-21:
Hi, 

I want to use the yun shield with the Arduino or Arduino mega ( I have both) and I was wondering if it would work just like an Arduino yun? 

Here's the stack overflow question: 
http://stackoverflow.com/q/34393144/5210362

deadprogram, 2015-12-29:
Hello, @manu354 

I have not tried this option, and do not have a Yun shield, although I do have a Yun. Probably you could follow the instructions we have for Yun? http://cylonjs.com/documentation/platforms/yun/ 

However, since the Yun shield does not appear to have an SD card, you will need to use a USB drive to transfer the files. In any case, please let us know how it goes if you decide to give it a try.

manu354, 2015-12-31:
The yun shield 2.3 has an mini sd  slot.

deadprogram, 2016-01-02:
That being the case it *should* work. Please let us know how it works out, if you give it a try.

manu354, 2016-03-18:
Hey, I just got it working with an arduino leonardo and yun shield. It works! It will work with any yun shield that has an sd card. Nothing to tweak, works exactly like an arduino yun.

closed 2016-03-18 by manu354