hybridgroup/cylon #324: Error when playing an audio file

sanoojnz, 2015-12-22:
Getting error when trying to play an audio file in response to an mpu6050 event..
work: function(my) {
         my.audio.play('cat.mp3');
}

The audio plays fine for the first event. Subsequent event throws the following error:

Error: write after end
  at writeAfterEnd (/home/pi/node_modules/cylon-audio/node_modules/lame/node_modules/readable-stream/lib/_stream_writable.js:144:12)
  at Decoder.Writable.write (/home/pi/node_modules/cylon-audio/node_modules/lame/node_modules/readable-stream/lib/_stream_writable.js:192:5)
  at ReadStream.ondata (_stream_readable.js:540:20)
  at ReadStream.emit (events.js:107:17)
  at readableAddChunk (_stream_readable.js:163:16)
  at ReadStream.Readable.push (_stream_readable.js:126:10)
  at onread (fs.js:1679:12)
  at FSReqWrap.wrapper [as oncomplete] (fs.js:527:17)

deadprogram, 2015-12-23:
Hi, @sanoojnz 

I'm actually thinking of switching the implementation of cylon-audio to use `node-mpg123` instead of the current `node-speaker` based implementation. It handles simultaneous playback a lot better, certainly.

Any thoughts on that?

sanoojnz, 2015-12-24:
Thanks Ron, for pointing me to mpg123.. It seems to work with the pi's
audio jack.. I am trying to get it to work with a bluetooth speaker
eventually.. Was trying node-web-audio-api library before. It looked
promising..

Sanooj

deadprogram, 2016-01-14:
Hi @sanoojnz I just released an update to https://github.com/hybridgroup/cylon-audio that uses MPG123. Please give some feedback when you get a chance. Thanks!

deadprogram, 2016-01-23:
Hi, @sanoojnz I am going to close this issue now, please reopen if you need further assistance with it, and thanks.

closed 2016-01-23 by deadprogram