hybridgroup/cylon #325: Anyone with examples or adaptor to use that with Makeblcok Orion o Mbot?

acacha, 2015-12-23:
Hello everyone and congratulations for your amazing job!

I haven't found any examples or references of how to use Cylon.js with Makeblock Orion or Mbot robot? Anyone could help me please?

deadprogram, 2015-12-23:
Hi, @acacha 

Since I do not have either of those devices, I took a quick look at their website.

The Orion should be able to be used with https://github.com/hybridgroup/cylon-firmata as are most Arduino compatible devices. 

To use the Mbot, you would need to have some additional software on the MBot itself running to handle the Bluetooth or WiFi communication. There are a number of examples of simple serial communication, actually cylon-rapiro does this. Serial over Bluetooth is the same thing, more or less. Wifi communication would require a bit more effort, there are several projects around that do this I think?

Anyhow, hope that helps!

acacha, 2015-12-26:
Hi @deadprogram thanks! Asap I will try your recommendations.