hybridgroup/cylon #327: 1-Wire DS18B20

rkelly92, 2016-01-07:
Hi,

I'm afraid the answer is no... but want to ask either way. I'm trying to use a DS18B20 temperature sensor.  From what I researched, ConfigurableFirmata is necessary to get this to work since it uses the 1-Wire bus protocol. Is there I way I could get this to work with Cylon?

deadprogram, 2016-01-07:
The cylon specific work would be adding the SPI interface to the cylon-firmata module, and then also creating a repo to contain drivers for any SPI devices (cylon-spi).

Adding SPI interfaces for other platforms would be a bonus, but this would lay the groundwork for more devices/platforms later.

rkelly92, 2016-01-08:
I'm rather new to working with this but isn't SPI different from 1-Wire? Either way, it's a bummer there's no support right now for it. The timeline from the project I'm working on may not allow me to spend too much time on this but I'll definitely give it a shot.

deadprogram, 2016-01-08:
Yes, I was responding a little too quickly, but same idea: the interface needs to be created in the adaptor, and then drivers can use that interface.

The defacto interface has been https://github.com/jgautier/firmata#one-wire so the first step would be adding that support to `cylon-firmata`.

rkelly92, 2016-01-11:
Still needs some work (better error handling, tests) but got it to work :smile: (https://github.com/rkelly92/cylon-one-wire, https://github.com/rkelly92/cylon-firmata/blob/firmata-one-wire/lib/firmata.js). Had one small problem though. I don't understand how the data property from the cylon Registry gets loaded. In the beginning, having cylon, cylon-firmata and my cylon-one-wire (which contains the driver) in the dependencies, the following error was thrown right away:
```
Cannot find the 'cylon-ds18b20' module.
This problem might be fixed by installing it with 'npm install cylon-ds18b20' and trying again.
```
The device was specified as:
```
devices: {
    sensor: { driver: 'ds18b20', pin: 2 }
  }
``` 
To get it working I initially added the module property to the device:
```
devices: {
    sensor: { driver: 'ds18b20', pin: 2, module: 'cylon-one-wire' } 
  }
```
Later on I found that I could also include cylon-one-wire to the dependencies array from the object exported by cylon-firmata, and that way I could omit the module property in the device definition. Bottom line is that it's not clear to me how to correctly establish/represent the relationship between the adaptor and the driver. How should this be done?

deadprogram, 2016-01-19:
Hi, @rkelly92 sorry somehow I overlooked your reply.

So the dependency concept is basically:

- modules can contain adaptors, drivers, or both.
- an adaptor-only module, can say "always load a set of drivers from some other module". This is useful when all of the drivers are known to work with the interfaces provided by that adaptor. An example of this you have already found is `cylon-firmata` always loading `cylon-gpio` & `cylon-i2c` drivers.
- a driver-only module, on the other hand, does not have to specify which adaptors it works with. It simply expects an interface be provided to the driver by the adaptor.
- a module with both adaptor & its own drivers is usually self-contained, such as the cylon-ardrone module.

So in your case, simply declaring which module the driver is in, allows the code to load as you expect. If you were to submit a PR on `cylon-firmata` that adds the 1-wire interface, then this interface can then also be added to other cylon adaptors in the future.

Anyhow, hopefully that clarifies things a bit. Thanks again for all your efforts on this.

rkelly92, 2016-03-26:
Hi @deadprogram. Thanks for your answer. Had some non-technical issues with my project that held back my work. I think my cylon-firmata fork is in a good state for a PR and receiving some feedback. To which branch should I submit the PR to?

deadprogram, 2016-03-26:
Hi @rkelly92 a PR against `master` please, and thank you.

deadprogram, 2016-04-26:
Hi @rkelly92 your PR is now released in cylon-firmata and also a link to your cylon-one-wire repo is in the main README for cylon. Thank you again for your contribution!

deadprogram, 2016-04-28:
Closing now that it has been released. Thanks again.

closed 2016-04-28 by deadprogram