hybridgroup/cylon #328: Using LeapMotion or keyboard and mip simultaneously doesn't work and produce error messages

lencorredera, 2016-01-13:
Hi, I'm trying to control a mip with a leap motion. There's no problem at all when I use them alone, but when I define a robot with both devices, I'm only able to receive/send data to the first defined one, and the other simply doesn't work. 

I tried to use keyboard instead of leap motion, but the problem seems to be the same, so I think it could be related to cylon-mip module or its dependencies.  

Please allow me to explain it with a couple of examples:

Case I:  leapmotion defined first (and mip in second position). 
I receive leap motion but I'm not able to send anything to the mip.

Code:
```
var Cylon = require('cylon');

Cylon.robot({
  connections:  { 
    leapmotion: { adaptor: 'leapmotion' },
    bluetooth:  { adaptor: 'central', uuid: '101xxxx7', module: 'cylon-ble'},
  },

  devices: 
  {
    leapmotion: { driver: 'leapmotion' },
    mip: {driver: 'mip'},
  },

  work: function(my) {
    my.mip.getUp(1);
    my.leapmotion.on('gesture', function(hand) {
      console.log(hand.type);
      my.mip.distanceDrive(0, 2, 0, 0);
    });
  }
}).start();
```
StackTrace:
```
/Users/user/node_modules/cylon-mip/lib/driver.js:322
  this.connection.writeServiceCharacteristic(s, c, new Buffer(value),
                  ^
TypeError: this.connection.writeServiceCharacteristic is not a function
    at Driver._writeServiceCharacteristic (/Users/luisenrique/node_modules/cylon-mip/lib/driver.js:322:19)
    at Driver.turnLeft (/Users/luisenrique/node_modules/cylon-mip/lib/driver.js:245:8)
    at Driver.<anonymous> (/Users/luisenrique/Dropbox/Tech/src/cylon-mip/leapmip.js:23:16)
    at emitOne (events.js:77:13)
    at Driver.emit (events.js:169:7)
    at Adaptor.<anonymous> (/Users/luisenrique/node_modules/cylon/lib/basestar.js:75:22)
    at emitOne (events.js:77:13)
    at Adaptor.emit (events.js:169:7)
    at Leap.loop.frame (/Users/luisenrique/node_modules/cylon-leapmotion/lib/adaptor.js:70:11)
    at emitOne (events.js:77:13)
```
Commenting my.mip lines will produce many leap motion gesture logs, as expected.

If I swap the order in connection and device definition (see code below), my mip gets up but I never get leap motion events.
```
Cylon.robot({
  connections:  { 
    bluetooth:  { adaptor: 'central', uuid: '101xxxx7', module: 'cylon-ble'},
    leapmotion: { adaptor: 'leapmotion' }
  },

  devices: 
  {
    mip: {driver: 'mip'},
    leapmotion: { driver: 'leapmotion' },
  },
```
Everything was happening on a Mac OS X with El Capitan up to date (v.10.11.2).

Thanks in advance for your support!

deadprogram, 2016-01-13:
Hi, @lencorredera 

You need to specify which device goes with which connection, like this:

```
var Cylon = require('cylon');

Cylon.robot({
  connections:  { 
    leapmotion: { adaptor: 'leapmotion' },
    bluetooth:  { adaptor: 'central', uuid: '101xxxx7', module: 'cylon-ble'},
  },

  devices: 
  {
    leapmotion: { driver: 'leapmotion', connection: 'leapmotion' },
    mip: {driver: 'mip', connection: 'bluetooth' },
  },

  work: function(my) {
    my.mip.getUp(1);
    my.leapmotion.on('gesture', function(hand) {
      console.log(hand.type);
      my.mip.distanceDrive(0, 2, 0, 0);
    });
  }
}).start();
```

Please give that a try.

lencorredera, 2016-01-13:
Hi @deadprogram, thank you for your incredibly fast response!

I tried your suggestion but it didn't work for the first time. Even more, bluetooth hardware went off. 
So I rebooted my computer and tried again. This time everything worked fine.

So thanks again for your support. You have been very helpful, @deadprogram!

closed 2016-01-15 by deadprogram