hybridgroup/cylon #330: How can I detect a device being connected

coprea, 2016-02-09:
This is more of a question not an issue, I hope I can get some help.

As I've seen on the blog, Cylon 1.2 has the ability to register devices later. In my case the device is a Digispark. After a lot of fiddling I figured I need to start the connection and then the device, so my code now looks like:
`digisparkDevice: function() {
    this.connection("digispark", { adaptor: "digispark" });
    this.startConnection(this.connections.digispark, function() {});

    this.device("red", {connection: "digispark", driver: "led", pin: 0});
    this.startDevice(this.devices.red, function() {});
    ... more led devices here...
  },`

Later in the work function I do:
`    my.digisparkDevice();
     every((1).second(), function() {
  	 my.red.toggle();
     });`
and it works great when the Digispark is connected. If not connected, I get a Segmentation fault. I tried toggle() in a try/catch but still the same error. I looked at the device object and didn't notice any difference between Digispark connected and not (I was trying to find if there is something undefined that I could check for).

I want to be able to check somehow if the Digispark is connected and if not just be able to use the other devices connected to my robot. Is this possible?

I'd appreciate any help. Thank you!

deadprogram, 2016-02-10:
Hi, @coprea try doing this (from your example):

```
if (this.connections.digispark.connected) {
   console.log("We are connected");
}
```

Hope that helps!

coprea, 2016-02-11:
This looks great, unfortunately .connected returns true even without the digispark connected so I run into the same issue. Thanks though, I didn't know about the .connected, maybe someday it will help.

deadprogram, 2016-02-11:
All `connected` currently indicates is that the `connect()` function was called, which apparently it was. 

What is the correct behaviour in your desired case, do you think? I would imagine something not unlike the BLE scanning example, except in this case waiting for an HID to be inserted.

coprea, 2016-02-11:
I am not sure, unfortunately I have no idea how to check the presence of a USB device. I will look at the BLE example you mentioned though and dig on the web. Thanks!

deadprogram, 2016-02-11:
According to https://github.com/hybridgroup/cylon-digispark/blob/master/lib/digispark.js#L38 this should be returning an error when there is no digispark plugged in. However, that code is either not detecting this case, or it being ignored upstream by cylon core.

I will dig thru my pile of boards to find a Digispark to see if I can try this out myself.