hybridgroup/cylon #332: REPL like?

bartmichu, 2016-03-12:
Hello, can I run Cylon programs in a way that enables me to fiddle and tweak functions/variables as it runs? I'm thinking something like ```repl.inject``` in Johnny-Five.

deadprogram, 2016-03-12:
Hi, @bartmichu it is very easy to use Cylon.js with https://github.com/maxogden/node-repl

```
$ node-repl demos/hello.js                                                                                             
> Cylon HTTP API server is now online.
Listening at https://127.0.0.1:3000

> Cylon.MCP.robots
{ test: 
   Robot {
     toJSON: [Function: bound ],
     connection: [Function: bound ],
     initRobot: [Function: bound ],
     device: [Function: bound ],
     start: [Function: bound ],
     startWork: [Function: bound ],
     startConnections: [Function: bound ],
     startConnection: [Function: bound ],
     startDevices: [Function: bound ],
     startDevice: [Function: bound ],
     halt: [Function: bound ],
     toString: [Function: bound ],
     log: [Function: bound ],
     setMaxListeners: [Function: bound setMaxListeners],
     getMaxListeners: [Function: bound getMaxListeners],
     emit: [Function: bound emit],
     addListener: [Function: bound addListener],
     on: [Function: bound addListener],
     once: [Function: bound once],
     removeListener: [Function: bound removeListener],
     removeAllListeners: [Function: bound removeAllListeners],
     listeners: [Function: bound listeners],
     listenerCount: [Function: bound listenerCount],
     name: 'test',
     running: false,
     connections: { loopback: [Object] },
     devices: { ping: [Object] },
     work: [Function],
     commands: {} } }
```

Hope that helps!

bartmichu, 2016-03-26:
Thanks it's exactly what I was looking for.

closed 2016-03-26 by bartmichu