hybridgroup/cylon #333: Cannot find cylon-gpio with arduino yun

manu354, 2016-03-14:
[image]

Hi,

I am trying to do simply turn a led off and one from an express website. This is my code for just the arduino - not the website code - 

Cylon.robot({
    connections: {
        arduino: { adaptor: 'firmata', port: '/dev/ttyATH0' }
    },

    devices: {
        led: { driver: 'led', pin: 13 }
    },

    work: function(my) {
        if (GLOBAL.led == false) {
            my.led.turnOn()
        }
        else if (GLOBAL.led == false) {
            my.led.turnOff()
        }
    }
}).start();

I have uploaded the firmata code to the arduino, and followed all the other steps.

manu354, 2016-03-14:
This the npm-debug.log 

```
root@dragino:/www/iot# vi npm-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
2 info using npm@1.4.28
3 info using node@v0.10.33
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info prestart iot@0.0.0
6 info start iot@0.0.0
7 verbose unsafe-perm in lifecycle true
8 info iot@0.0.0 Failed to exec start script
9 error iot@0.0.0 start: `node ./bin/www`
9 error Exit status 143
10 error Failed at the iot@0.0.0 start script.
10 error This is most likely a problem with the iot package,
10 error not with npm itself.
10 error Tell the author that this fails on your system:
10 error     node ./bin/www
10 error You can get their info via:
10 error     npm owner ls iot
10 error There is likely additional logging output above.
11 error System Linux 3.3.8
12 error command "/usr/bin/nodejs" "/usr/bin/npm" "start"
13 error cwd /www/iot
14 error node -v v0.10.33

```

deadprogram, 2016-03-14:
Hi, @manu354 you are using `npm start` which is the script to start express.js not cylon.js

To run cylon.js just use the command `node <myprogram.js>` instead.

Hope that helps!

manu354, 2016-03-14:
If I were to use node app I would get the same error.

My cylon code is built into my epress app.js

manu354, 2016-03-14:
I'll try running the cylon app without any of the express files, but to me it doesn't seem like the error has anything to do with that?

deadprogram, 2016-03-14:
Did you try running `npm install cylon-gpio` as it requests in your first error log?

manu354, 2016-03-15:
Thanks!

Doing `npm install cylon-gpio cylon-i2c` on my pc then transefering to the yun fixed it!

closed 2016-03-15 by manu354