hybridgroup/cylon #334: How can I automatically start my cylon program on the arduino yun.

manu354, 2016-03-18:
Hey,

I'm building a project where my arduino will reboot regularly and I need the same node program to be run on boot every time.

I need to run `node mnt/sda1/arduino/www/iot/bin/www.js`

On the openwrt website it said I can do this. https://wiki.openwrt.org/inbox/procd-init-scripts : 

    #!/bin/sh /etc/rc.common
    
    USE_PROCD=1
    
    start_service() {
      procd_open_instance
      procd_set_param command node /mnt/sda1/arduino/www/iot/bin/www.js
      # respawn automatically if something died, be careful if you have an alternative process supervisor
      # if process dies sooner than respawn_threshold, it is considered crashed and after 5 retries the service is stopped
      procd_set_param respawn ${respawn_threshold:-3600} ${respawn_timeout:-5} ${respawn_retry:-5}
      procd_close_instance
    }

However i'm not sure what who to exactly write it as nothing comes up when I try run it with `/etc/init.d/node-app start` I am obviously writing the code wrong but i'm not sure what it should exactly look like, It would be great if someone could provide an example.

I have asked this on stack overflow & arduino forums and not gotten a reply. Sorry to bother you with this question!

Thanks!

manu354, 2016-03-18:
I found an example but don't know how to use it for my use.

	#!/bin/sh /etc/rc.common
	# Copyright (C) 2013-2014 OpenWrt.org
	
	START=50
	USE_PROCD=1
	
	start_service() {
	        procd_open_instance
	
	        procd_set_param command /usr/bin/xupnpd
	        procd_append_param command -d /usr/share/xupnpd
	
	        procd_set_param respawn
	        procd_close_instance
	}

deadprogram, 2016-03-18:
According to the docs I found, you would need to run that script using this command:

```
/etc/init.d/node-app start_service
```

manu354, 2016-03-18:
Thanks I'll try that.

manu354, 2016-03-19:
That''s not a command. 

```
root@dragino:/etc/init.d# /etc/init.d/node-app start_service
Syntax: /etc/init.d/node-app [command]

Available commands:
        start   Start the service
        stop    Stop the service
        restart Restart the service
        reload  Reload configuration files (or restart if that fails)
        enable  Enable service autostart
        disable Disable service autostart
```

deadprogram, 2016-03-19:
My comment applied to the script you pasted the text of in the first example (https://github.com/hybridgroup/cylon/issues/334#issue-141996156), let's us call it `yourscript`. You need to type:

`/path/to/yourscript start_service`

manu354, 2016-04-04:
Fixed by just adding the command to rc.local

closed 2016-04-04 by manu354