hybridgroup/cylon #335: I'd like to add support for an ES6 syntax

ringzhz, 2016-03-19:
Hello, I'd like to work with you to add es6 class-based syntax for driver/robot definitions.
Have you already considered/done it?

deadprogram, 2016-03-19:
Hi, @ringzhz considered yes, but implemented, not yet.

An ES6 class based syntax for adaptors and/or drivers would be fantastic.

We also need to write some examples of robots using ES6 syntax, which is quite simple to do.

In any case, your contributions would be greatly appreciated!

deadprogram, 2016-03-20:
As a simple example of a robot with ES6 syntax:

```javascript
let Cylon = require("cylon");

Cylon.api()

Cylon.robot({
  name: "test",

  connections: {
    looper: { adaptor: "loopback" }
  },

  devices: {
    pinger: { driver: "ping" }
  },

  work() {
    every((1).seconds(), () => {
      console.log("Hello, human!");
      console.log(this.pinger.ping());
    })

    after((5).seconds(), () => console.log("It has been 5 seconds"))
  }
})

Cylon.start()
```

ringzhz, 2016-03-20:
That's cool. I think that getting to that place would be fairly easily achievable with the use of a transpiler. 

I was thinking something a little more OO like:

```javascript
import {Robot} from 'cylon';

export class DavesRobot extends Robot {
  // name may be an argument to Robot's constructor?
  getConnections() {
    return {
      looper: { adaptor: "loopback" }
    };
  }
  getDevices() {
    return {
      pinger: { driver: "ping" }
    };
  }
  work() {
    // ...
  }
}
```

This may require some modifications to the way your code is set up. I haven't looked through your source yet, but do you think that would be a worthwhile improvement?

deadprogram, 2016-03-20:
The place that sort of thing becomes much more useful, IMO, is the creation of adaptors/drivers, where they already are using some inheritance by design.

ringzhz, 2016-03-20:
Absolutely, and I just figured it would make sense to follow a similar pattern in the robot code, but indeed, the drivers are where I think this change would make the biggest improvement to code readability/terseness.

deadprogram, 2016-03-20:
The place in the robot code where ES6 would really shine, would be using Promises to control the sequence of events, especially those triggered by `every` and `after`. But indeed, other opportunities for syntactical sugar may become apparent.

So you are "in"? :smile:

ringzhz, 2016-03-20:
For sure. Glad we're on the same page. I'll start taking a look at your codebase when I get a chance and see if I can get a feel for it. Thanks for the prompt responses!