hybridgroup/cylon #336: User modifiable properties

bartmichu, 2016-04-02:
Hi, would you consider adding (or specifying) user modifiable field on devices for the purpose of storing non-standard properties?

I find myself adding extra functionality to some devices quite often (e.g. simple hold event handler on a button):
```
work: function (my) {
    var holdTime = 2000;
    var timeout = null;
    
    function holdHandler() {
      console.log("hold");
    }

    my.button.on("push", function () {
      console.log("push");
      timeout = setTimeout(holdHandler, holdTime);
    });
    
    my.button.on("release", function () {
      console.log("release");
      if (timeout && !timeout._called) {
        clearTimeout(timeout);
      }
    });
  }
```

It begs moving ```holdTime``` and ```timeout``` to the button:
```
devices: {
  button: {
    driver: "button",
    pin: 4,
    userProperties: {
      holdTime: 200,
      timeout: null
    }
  }
}
```

I can do that right now and it works fine, but I don't really know if it's going to break something down the road.
So the idea is: you specify some field to users like "put your stuff in here" and similarly to module developers "hey don't touch this".

Does it make sense? It's about convention I guess.

deadprogram, 2016-04-04:
Hi, @bartmichu 

Thanks for the thoughtful question. Exactly, there are 2 ways to add "fields" to your robot, either by creating some var that is in scope for your needs, or else a property on the robot object itself. the vars declared within the `work` function will be in scope for any other functions called while the robot is running. But it is a little "messy" perhaps. I typically instead just add them to my robot object itself, which seems cleaner.

Your example showing the `userProperties` added to the device is a similar but different need. This already happens right now with any properties not reserved by Cylon.js itself: https://github.com/hybridgroup/cylon/blob/master/lib/driver.js#L41

Anyhow, I think the solution is probably to better document this behaviour, rather than add any code. What do you think?

bartmichu, 2016-04-04:
Thank you for the detailed and informative response. Adding directly on the robot makes sense, I was not aware of reserved property checks.

I guess it could be documented a bit better :)

closed 2016-04-04 by bartmichu