hybridgroup/cylon #337: Espruino

EthraZa, 2016-04-03:
Hi.
Is it possible to use CylonJS within Espruino Pico? 
If no, there are plans to support Espruino?

deadprogram, 2016-04-03:
Hi, @EthraZa I have a couple of them, but have not had a chance to work with them yet. But other than the memory limits, seems like it is certainly possible.

EthraZa, 2016-04-03:
Hi @deadprogram, thanks for answering.
I have asked because I'm old JS developer but a noob in the maker scene. I'm still planning to buy my first Arduino kit and when I saw this Espruino Pico that could run JS on their own without a PC controlling it and can be attached to an Arduino adaptor board, my eyes have shined.

Here in my contry the only thing I can find to buy is Arduino UNO copies and some of it's shields. I was hoping to be able to just port any code I make to Arduino over to Espruino in the future, to get little projects running standalone.

I doubt I have the skills or the time to acquire them to port CylonJS to Espruino, but I still hope some one can do it. :)