hybridgroup/cylon #338: Missing pin number in log output for pin 0

mortenf, 2016-04-20:
In Robot.startDevice we have the following:

```javascript
  var str = "Starting device '" + device.name + "'";

  if (device.pin) {
    str += " on pin " + device.pin;
  }
```

However, for devices connected to pin 0 (analog pin common on Arduinos), the log output doesn't say which pin number it is.

It should probably be something like this instead:

```javascript
  var str = "Starting device '" + device.name + "'";

  if (device.pin || device.pin === 0) {
    str += " on pin " + device.pin;
  }
```

I'd be happy to provide a PR, if interested.

deadprogram, 2016-04-20:
Hi @mortenf that makes sense to me. A PR would be very much appreciated!

deadprogram, 2016-05-16:
Hi, @mortenf ever have a chance to work on this? :)

mortenf, 2016-05-21:
Finally. Sorry for the delay.

deadprogram, 2016-08-26:
Thanks @mortenf this was merged a whole ago. Closing this issue.

closed 2016-08-26 by deadprogram