hybridgroup/cylon #340: How to make cylon support to control an arduino-based robotic arm in real time?

iszufu, 2016-04-25:
Here is a library written in JavaScript for controlling an arduino-based robotic arm [uarmforjs](https://www.npmjs.com/search?q=uarmforjs). Currently, it can work very well in node. But I wish to control it in a chrome app, which will definitely need more dependencies, browserify for example, according to your official documents. On the other hand, I hope to interact with the robotic arm in real time in chrome app. I think to meet this demanding the cylon API (http or mqtt) might be used, but I am not sure, because I don't really know much about cylon framework.
I think there should be some work to be done to achieve these, to make it easier for me to reach the end, can you please give me some relatively specific ideas? If an example which is similar to this can be linked, that would be perfect.
Thanks.

deadprogram, 2016-04-25:
Hi @iszufu 

I was checking out the firmware there, and it appears to be a hand-customized version of Firmata.

First of all, you can use Cylon.js with Firmata via the cylon-firmata module (https://github.com/hybridgroup/cylon-firmata)

Secondly, as a suggestion, rather than just modifying Firmata itself, the UArm use case case is exactly what ConfigurableFirmata was created for. Check it out at https://github.com/firmata/ConfigurableFirmata

Since the UArm's EEPROM implementatation is already in the form of SYSEX messages, it would probably be pretty easy to changeover to use the plugin approach, and hence be able to pickup upstream updates to Firmata itself. Anyhow, not sure if you are one of the maintainers of that, but just a suggestion.

Hopefully that helps get you started.

deadprogram, 2016-04-25:
Tp elaborate on this a little further, please check out the example at https://github.com/hybridgroup/cylon-firmata/blob/master/examples/curieIMU/curieIMU.js which is probably what you are looking for, as far as Cylon.js examples that communicates in real-time, and also uses both the cylon-gpio drivers, along with custom SYSEX commands, both at the same time.

iszufu, 2016-05-13:
Hi, @deadprogram , firstly I would like to thank you for your last help, really appreciate it. This time I want to ask something about cylon.js in PhoneGap app. I have seen the [PhoneGap + cylon.js ](https://github.com/hybridgroup/phonegap-cylon-spark) example on github, but I prefer to control an arduino board with BluetoothSerial from a PhoneGap app, rather than spark. Can you please show me some examples if someone has done this, or give some ideas to achieve this if no one has tried this.