hybridgroup/cylon #341: Events.js 141 - Unhandled "error" event

mrojecki, 2016-05-14:
I don't know why, but I'm getting error when I'm trying to use cylon.js

I have RPI 3 with newest Raspbian and Node.js 4.4.4.

I'm trying to run this code to run electric motor attached to PCA9685 custom RPi Hat with this code:

`var Cylon = require("cylon");

Cylon.robot({
    connections: {
        raspi: {
            adaptor: 'raspi'
        }
    },

    devices: {
        pca9685: {
            driver: 'pca9685'
        }
    },

    work: function(my) {

        my.pca9685.setPWMFreq(60);

        my.pca9685.setPWM(2, 0, 1000);
        my.pca9685.setPWM(3, 4096, 0);
        my.pca9685.setPWM(4, 0, 0);

    }
}).start();`

`pi@raspberrypi:~/Desktop/Raspby $ node app.js
2016-05-14T00:35:19.170Z : [Robot 1] - Starting connections.
2016-05-14T00:35:19.217Z : [Robot 1] - Starting connection 'raspi'.
2016-05-14T00:35:19.225Z : [Robot 1] - Starting devices.
2016-05-14T00:35:19.226Z : [Robot 1] - Starting device 'pca9685'.
2016-05-14T00:35:19.232Z : [Robot 1] - Working.
events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: EIO, Input/output error
    at Error (native)
`

I've got all NPM modules and I've checked documentation twice.

Simmilar code written in Python works:

`import Adafruit_PCA9685
 
pwm = Adafruit_PCA9685.PCA9685(address=0x60)
pwm.set_pwm_freq(1000)
 
pwm.set_pwm(2, 0, 1000)
pwm.set_pwm(3, 4096, 0)
pwm.set_pwm(4, 0, 0)`

Any help will be appericiated.

deadprogram, 2016-05-14:
Hi @mrojecki 

I noticed that your Python code is using i2c address 0x60 and the cylon-i2c driver is using 0x40 as the default here: https://github.com/hybridgroup/cylon-i2c/blob/master/lib/pca9685.js#L24

Can you please try changing this code in your sample program:

```javascript
pca9685: {
  driver: 'pca9685'
}
```

to this:

```javascript
pca9685: {
  driver: 'pca9685', address: 0x60
}
```

Give that a try, and see if that helps.

mrojecki, 2016-05-14:
How I could not seen that...
Oh, well. 
Sir, You are lifesaver :)

deadprogram, 2016-05-15:
Glad to be of assistance. I will close this issue, please reopen if needed.

closed 2016-05-15 by deadprogram