hybridgroup/cylon #344: Error thrown when sending second start command to cylon

karltaylor, 2016-06-10:
Using Node.js, express and cylon.js, I'm getting a post request and blinking an LED using a raspberry Pi. I receive the first post request fine, the code runs and the LED blinks. However on the second post request I receive an error:

```javascript
events.js:146
      throw err;
      ^

Error: Uncaught, unspecified "error" event. (Error occurred while writing value 1 to pin 23)
    at DigitalPin.emit (events.js:144:17)
    at DigitalPin.<anonymous> (/home/pi/tech-corner/devTracker/node_modules/cylon/lib/io/digital-pin.js:70:12)
    at fs.js:1134:21
    at FSReqWrap.oncomplete (fs.js:82:15)
```

I'm using pin: 16. So I'm confused as to why it's trying to write to pin 23.

```javascript
var express = require('express')
var bodyParser = require('body-parser')
var app = express()
var Cylon = require('cylon')

var pi = Cylon.robot({
  connections: {
    raspi: { adaptor: 'raspi' }
  },

  devices: {
    led: { driver: 'led', pin: 16 }
  },

  work: function (my) {
    after((1).second(), function () {
      console.log('Post request recieved! Toggle the LED')
      my.led.toggle()
    })

    after((2).second(), function () {
      my.led.toggle()
    })

    after((5).seconds(), function () {
      console.log('I\'m shutting down now.')
      pi.halt()
    })
  }
})

app.set('port', process.env.PORT || 1337 - 1)

// parse application/x-www-form-urlencoded
app.use(bodyParser.urlencoded({ extended: false }))

// parse application/json
app.use(bodyParser.json())

app.post('/', function (req, res) {
  console.log(req.body.user_name)

  pi.start()

  res.end('Hello')
})

app.listen(app.get('port'))
```

deadprogram, 2016-08-26:
Just out of curiousity, any specific reason to not just use the https://github.com/hybridgroup/cylon-api-http plugin?

devsheder, 2016-09-24:
Hi,
I have the same error as above but without using express.

I want to use one single instance of Cylon robot to control LED.
I store the result of Cylon.robot({...}) in a global variable in my node script. I can start() then halt() the robot but when I try to call start() method again I have this error thrown :
events.js:146
      throw err;
      ^
Error: Uncaught, unspecified "error" event. (Error occurred while writing value 1 to pin 17)
    at DigitalPin.emit (events.js:144:17)
    at DigitalPin.<anonymous> (/home/pi/leo/leo_bt_service/node_modules/cylon/lib/io/digital-pin.js:70:12)
    at fs.js:1139:21
    at FSReqWrap.oncomplete (fs.js:82:15)

I don't use the 17 PIN but only those pin :
devices: {
			led1: { driver: 'led', pin: 11},
			led2: { driver: 'led', pin: 13},
			led3: { driver: 'led', pin: 15},
			led4: { driver: 'led', pin: 12},
			led5: { driver: 'led', pin: 16},
			led6: { driver: 'led', pin: 18}
		}

Do you have an idea please ?

Thanks,
Regards.

spacecowboyian, 2017-08-08:
Ever figure this one out? I'm just digging into a motor project and I'm getting the same error.

karltaylor, 2017-08-08:
@spacecowboyian Not from Cylon, I decided to use [pigpio](https://github.com/fivdi/pigpio) which worked fine!

MiguelAngel82, 2017-08-13:
Hi all,

Similar to this issue I have problems running the simple example of blinking LED on a Raspberry Pi.

I've created the simple example, called index.js with this code:

```
var Cylon = require("cylon");

Cylon.robot({
  connections: {
    raspi: { adaptor: 'raspi' }
  },

  devices: {
    led: { driver: 'led', pin: 11 }
  },

  work: function(my) {
    every((1).second(), my.led.toggle);
  }
}).start();
```

I've installed NVM and then I've installed Node v6.9.0. After that, I've installed all this modules:
```
npm install cylon
npm install cylon-raspi
npm install cylon-firmata
npm install cylon-gpio
npm install cylon-i2c
```
Finally, I've executed Node to run the example, like this:
`node index.js`

When this command is executed at first time this result is obtained:

```
pi@raspberrypi:~/projects/led-example$ node index.js
2017-08-13T20:14:25.484Z : [Robot 1] - Starting connections.
2017-08-13T20:14:25.692Z : [Robot 1] - Starting connection 'raspi'.
2017-08-13T20:14:25.727Z : [Robot 1] - Starting devices.
2017-08-13T20:14:25.731Z : [Robot 1] - Starting device 'led' on pin 11.
2017-08-13T20:14:25.733Z : [Robot 1] - Working.
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. (Setting up pin direction failed)
    at DigitalPin.emit (events.js:163:17)
    at DigitalPin._setModeCallback (/home/pi/node_modules/cylon/lib/io/digital-pin.js:152:17)
    at DigitalPin.<anonymous> (/home/pi/node_modules/cylon/lib/io/digital-pin.js:146:10)
    at fs.js:1305:7
    at FSReqWrap.oncomplete (fs.js:123:15)
```

However, when this command is executed at second time (after the first run), everything works fine.

Any ideas? 

Thank you!

Regards,
Miguel.

mymattcarroll, 2017-11-07:
@MiguelAngel82, I am getting the same error every second time running a very similar script. Were you able to resolve this issue?

pipegreyback, 2017-12-25:
Same issue here. When the script runs for the second time, for some reason, it works.