hybridgroup/cylon #347: How to register just created new driver

tamer1an, 2016-07-24:
I generate a driver using cylon-cli

How to register it for futher usage?

As i understand i need to instal it as npm module? i did it. 

Has installed a just generated driver using `npm i  ./mydriver`

than i require it in my program 
```javascript
var Cylon = require('cylon');
var ultrasonic = require('mymodule);
```

and i get Cannot find the 'cylon-mymodule' module.
This problem might be fixed by installing it with 'npm install cylon-ultrasonic' and trying again.

cand find any instruction on that topic here:
https://cylonjs.com/documentation/adding-new-hardware-support/

and irc is also silent 

please help

Regards
Andrii

deadprogram, 2016-07-25:
Hi @tamer1an 

You are correct, there is some important info missing from the "New Hardware Support" page.

When implementing a module that includes Cylon.js drivers, you need to export a `drivers` function, and a `driver` function.

`drivers()` should return all of the drivers contained within that module, for example https://github.com/hybridgroup/cylon-i2c/blob/master/index.js#L25

`driver()` should return a new Driver object for a specific driver, just like https://github.com/hybridgroup/cylon-i2c/blob/master/index.js#L27-L33

Cylon.js will try to auto-require a module named `cylon-drivername` where "drivername" is the name of the driver. If it can require this specially named module, it will try to call the `driver()` function to instantiate a new Driver of the desired type.

The same is also true about "Adaptors", but you were specifically asking about Drivers.

Hope this helps!

deadprogram, 2016-07-26:
I've added an issue on the Cylon.js website to update that page: https://github.com/hybridgroup/cylon-site/issues/124

tamer1an, 2016-07-26:
Hello @deadprogram 

My driver was written correctly, before open this Issue I has investigate different examples in src-code and after your answer now I am  sure that is written it correctly. 

But here what was registration problem that i want to solve. Your answer above also gave me i insight  so i check that driver is working just fine but.

Here is the issue:
Take a look at this project [structure](https://github.com/tamer1an/home-defence): 

as always [package.json](https://github.com/tamer1an/home-defence/blob/master/package.json) has all dep
```javascript 
{
"dependencies": {
    "cylon": "^1.3.0",
    "cylon-firmata": "^0.24.0",
    "cylon-gpio": "^0.29.0",
    "cylon-i2c": "^0.26.1"
  }
}
```
e.g. I install dependencies than i want to install my custom driver for now all custom is in folder 
[drivers ](https://github.com/tamer1an/home-defence/tree/master/drivers)

here is the [index.js](https://github.com/tamer1an/home-defence/blob/master/drivers/ultrasonic/index.js)
/drivers/ultrasonic/lib/ultrasonic.js
and the [driver ](https://github.com/tamer1an/home-defence/blob/master/drivers/ultrasonic/lib/ultrasonic.js)

here is the actual [program ](https://github.com/tamer1an/home-defence/blob/master/src/js/ultrasonic.js) that i am running 

so i made following steps:
1) `$ npm i`
2)` $ npm i ./drivers/ultrasonic`
3)` $ node src/js/ultrasonic`

4) and i get _Cannot find the 'cylon-mymodule' module.
This problem might be fixed by installing it with 'npm install cylon-ultrasonic' and trying again._

if i add my [driver ](https://github.com/tamer1an/home-defence/blob/master/drivers/ultrasonic/lib/ultrasonic.js) here:  .\node_modules\cylon-i2c\lib\ultrasonic.js

and add it to the index.js file .\node_modules\cylon-i2c\index.js
```javascript
var Drivers = {
  // ...
  "ultrasonic": require("./lib/ultrasonic")
};
```

the problem is solved no error driver is working, how can i do the same(register my custom driver to work in the Cylon program) but without adding it to node_modules source folder?

closed 2016-07-30 by tamer1an