hybridgroup/cylon #359: I feel like an idiot - Compile? Deploy?

GantMan, 2017-01-01:
Just got started with CylonJS, and wrote JavaScript blink.  Yayyyy.

I like the REPL style interface of how it stays connected over firmata etc.

But I just want to disconnect it from my computer.  I can't seem to figure out or find any info on this, saying if it's possible.

Do I compile my JS to a HEX that I can upload via gort?
Do I bundle my JS like React Native?
Is it just impossible, but no one has asked or begged you to document this before?

Sorry for such a simple/dumb question, but I can't be the first person to ask this.

When I use the Arduino app for blink, it writes the file and TADA! no computer needed.  What am I missing?

deadprogram, 2017-01-02:
Hi @GantMan if you need to run completely independently on your microcontroller, than JS is probably not for you, since the JS interpreter is not running on the board itself. This is the case anytime you are using Firmata with any microcontroller, you always need some outside software to control it. Basically you are using the microcontroller as a peripheral.

The one exception to this those that implement some subset of JS interpreters such as used by http://www.espruino.com/

Hope that helps!

GantMan, 2017-01-02:
Ahhh!  I was wondering!  Thank you so much for your response.  I think it would be advantageous to identify that right out for Cylon users, since it breaks away from standard Arduino structure.  Kind of renders my use of Cylon for my friend's project, useless :(  as he needs it to be mobile.

I always am interested in the concept of bundling the javascript like React Native does OR compiling to HEX, which shouldn't be hard with some kind off LLVM?

In any case, what an amazing project!  I hope I find use for it soon :)

deadprogram, 2017-01-09:
Thanks @GantMan closing this issue now.

closed 2017-01-09 by deadprogram