hybridgroup/cylon #370: Reference: my is not defined

Pazox, 2017-07-15:
I am trying to use Cylon.js with my arduino, and I got stuck with this problem. the code I use:
```
var Cylon = require('cylon');

Cylon.robot({
  connections: {
    arduino: { adaptor: 'firmata', port: 'COM3' }
  },

  devices: {
    led: { driver: 'led', pin: 2}
  },

   work: function() {
    every((1).second(), function() {
        my.led.toggle();
    });
  }
});
Cylon.start();
```
the error I get:
```
        my.led.toggle();
        ^

ReferenceError: my is not defined
```
I tried to switch 'my' with 'this' and the error I get is: 
```
        this.led.toggle();
                ^

TypeError: Cannot read property 'toggle' of undefined
```
Any suggestions? Thanks in advance.

vkarpuram, 2017-07-20:
Modify your work function like so:

 work: function(my) {
    every((1).second(), function() {
        my.led.toggle();
    });
  }

Pazox, 2017-07-21:
That worked! Thanks.

closed 2017-07-21 by Pazox